picolibc/picolibc #800: sys/time.h no long includes time.h breaking some external code bases

sgstreet, 2024-08-05:
[include: Verify that all C headers expose only defined symbols ](https://github.com/picolibc/picolibc/commit/5e4d0c80f49b4efe8be85e36b5e85d65bd343214) removes, from sys/time.h, the inclusion of time.h.  This code is present in newlib and some external code [pico-sdk runtime.c](https://github.com/raspberrypi/pico-sdk/blob/6a7db34ff63345a7badec79ebea3aaef1712f374/src/rp2_common/pico_runtime/runtime.c#L239) depend on this inclusion.  Admittedly, this is problematic behavior on the part of the pico-sdk, but there may be other dependencies like this in the wild.  Is the divergence from the newlib implementation desirable?

keith-packard, 2024-08-05:
Just trying to match the specs more closely. I spent a bit of time trying to stop nesting includes of global headers where the spec says "may make visible" as that seems sloppy; I think it's cleaner to have each header file only define what the POSIX or C spec says they must. 

https://pubs.opengroup.org/onlinepubs/007904875/basedefs/sys/time.h.html

In this case, `sys/time.h` is allowed to make the symbols from `sys/select.h` visible, but it is not explicitly allowed to make the symbols from `time.h` visible. Of course, `sys/select.h` is allowed to make the symbols from `time.h` visible, but the spec is not clear whether that also allows `sys/time.h` to make those visible. Right now, picolibc's `sys/time.h` does happen to include `sys/select.h`, which is something I'd like to fix. But, `sys/select.h` doesn't include `time.h`, which is required to make `sys/time.h` conform to the explicitly written parts of the POSIX spec.

It should be easy enough to make pico-sdk's runtime.c work with picolibc by simply including the POSIX header files which define all of the symbols that it uses. In this case, you'd need to include `time.h` where `CLOCKS_PER_SEC` will be defined. That should work fine with newlib as well.

keith-packard, 2024-08-07:
> Have you considered passing the time.h removal to the newlibc maintainers?

This definitely seems like a change they wouldn't want -- newlib has a much stronger backward compatibility requirement than picolibc, even when the library is not in strict conformance with specifications, especially for Cygwin targets. Picolibc's embedded environment focus means it gets exposed to code designed to work with numerous other embedded toolchains. In that world, strict standards conformance is the best way I know to avoid problems.

closed 2026-11-10 by keith-packard