picolibc/picolibc #803: (Ease) usage with clang

nmeum, 2024-08-08:
The [picolibc usage instructions](https://github.com/picolibc/picolibc/blob/main/doc/using.md) recommend using picolibc via [the provided `.spec` file](https://github.com/picolibc/picolibc/blob/main/picolibc.specs.in). Unfortunately—as you are probably well aware—[clang doesn't support GCC's `.spec` file format](https://clang.llvm.org/docs/DriverInternals.html#relation-to-gcc-driver-concepts). I was therefore wondering what the recommended way of using picolibc with clang (instead of gcc) is. As an example, the embedded operating system [RIOT](https://riot-os.org) [currently links all code with GCC](https://github.com/RIOT-OS/RIOT/blob/2024.07/makefiles/toolchain/llvm.inc.mk#L15-L20), due to the lack of `.spec` support in clang). Which is rather hacky and undesirable as RIOT therefore still requires a gcc cross-compiler toolchain, and hence doesn't fully benefit from clang's more straightforward cross-compilation setup.

Recently(?), clang gained support for [configuration files](https://clang.llvm.org/docs/UsersManual.html#configuration-files) which—similar to GCC's `.spec` files—allow grouping “command-line options and allow all of them to be specified just by referencing the configuration file”. I haven't looked into it in greater detail yet, but maybe utilization of picolibc with clang could be eased significantly by providing such a configuration file in addition to the [existing `.spec` file](https://github.com/picolibc/picolibc/blob/main/picolibc.specs.in)? This would, for example, significantly ease utilization of picolibc with clang in the aforementioned RIOT operating system.

If there is interest in that I can look into coming up with a PR. If there is an easy existing solution for using picolibc in conjunction with clang that I missed so far, then please let me know :)

badcf00d, 2024-08-08:
Don't know enough to help directly, but I know that Arm's LLVM toolchain uses picolibc, and (as far as I'm aware) compiles it with clang as part of the build process. So you could perhaps take a look at what they're doing there. The steps for building from source are fairly straight forward:

https://github.com/ARM-software/LLVM-embedded-toolchain-for-Arm/blob/main/docs/building-from-source.md

keith-packard, 2024-08-08:
> If there is any interest in this, I can look into coming up with a PR. If there is an easy existing solution for using picolibc in conjunction with clang that I missed so far, then please let me know :)

Sounds like a very useful feature. 

 * Could we auto-generate this file from meson and cmake?
 * One of the tricks with the picolibc.specs file is that it gets installed into the GCC bits so you don't need to provide a complete pathname when using it. Could we make this work with clang?
 * Would this support the printf/scanf configuration stuff?

nmeum, 2024-08-09:
> * One of the tricks with the picolibc.specs file is that it gets installed into the GCC bits so you don't need to provide a complete pathname when using it. Could we make this work with clang?

Yes, this is definitively possible.

From https://clang.llvm.org/docs/UsersManual.html#configuration-files

> Configuration files can be either specified on the command line or *loaded from default locations*. […] If the provided argument contains a directory separator, it is considered as a file path, and options are read from that file. Otherwise the *argument is treated as a file name and is searched for sequentially in the directories*:
>
> * user directory,
> * system directory,
> * the directory where Clang executable resides.

Just as with the GCC `.spec` files, distributors will then just have to install the file into the clang system directory (CMake parameter `CLANG_CONFIG_FILE_SYSTEM_DIR`).