picolibc/picolibc #805: Incorrect detection of builtin functions for clang.

niedzwiecki-dawid, 2024-08-13:
Build system detects builtin functions incorrectly for clang:
```
  -
    kind: "try_compile-v1"
    backtrace:
      - "/mnt/host/source/src/third_party/zephyr/picolibc/cmake/picolibc.cmake:151 (try_compile)"
      - "/mnt/host/source/src/third_party/zephyr/picolibc/CMakeLists.txt:177 (picolibc_flag)"
    directories:
      source: "/mnt/host/source/src/platform/ec/twister-out/native_sim/tests/lib/cpp/cxx/cpp.main.picolibc/modules/picolibc/CMakeFiles/CMakeTmp"
      binary: "/mnt/host/source/src/platform/ec/twister-out/native_sim/tests/lib/cpp/cxx/cpp.main.picolibc/modules/picolibc/CMakeFiles/CMakeTmp"
    cmakeVariables:
      CMAKE_CXX_COMPILER_TARGET: "x86_64-pc-linux-gnu"
      CMAKE_C_COMPILER_TARGET: "x86_64-pc-linux-gnu"
      CMAKE_C_FLAGS: ""
      CMAKE_C_FLAGS_DEBUG: "-g"
      CMAKE_EXE_LINKER_FLAGS: ""
      CMAKE_MODULE_PATH: "/mnt/host/source/src/third_party/zephyr/main/cmake/modules"
    buildResult:
      variable: "_HAVE_BUILTIN_ISINFL"
      cached: true
      stdout: |
        Change Dir: /mnt/host/source/src/platform/ec/twister-out/native_sim/tests/lib/cpp/cxx/cpp.main.picolibc/modules/picolibc/CMakeFiles/CMakeTmp
        
        Run Build Command(s):/usr/bin/ninja -v cmTC_5e6a5 && [1/2] /usr/bin/x86_64-pc-linux-gnu-clang --target=x86_64-pc-linux-gnu   -Werror=missing-declarations -Werror=unknown-attributes -Werror=attributes -Werror=unsupported-floating-point-opt -Werror=ignored-optimization-argument -Werror=attribute-optimize -Wno-unused-command-line-argument -MD -MT CMakeFiles/cmTC_5e6a5.dir/have-builtin-isinfl.c.obj -MF CMakeFiles/cmTC_5e6a5.dir/have-builtin-isinfl.c.obj.d -o CMakeFiles/cmTC_5e6a5.dir/have-builtin-isinfl.c.obj -c /mnt/host/source/src/third_party/zephyr/picolibc/cmake/have-builtin-isinfl.c
        /mnt/host/source/src/third_party/zephyr/picolibc/cmake/have-builtin-isinfl.c:2:29: warning: use of unknown builtin '__builtin_isinfl' [-Wimplicit-function-declaration]
            2 | int main(void) { return foo(__builtin_isinfl((long double)42)); }
              |                             ^
        1 warning generated.
        [2/2] : && /usr/bin/cmake -E rm -f libcmTC_5e6a5.a && /usr/bin/llvm-ar qc libcmTC_5e6a5.a  CMakeFiles/cmTC_5e6a5.dir/have-builtin-isinfl.c.obj && /usr/bin/llvm-ranlib libcmTC_5e6a5.a && :
        
      exitCode: 0
```

Despite the fact bulitin function is unknown, the exitCode is still 0.

Adding `-Werror=implicit-function-declaration` to the [_picolibc_supported_compile_options](https://github.com/picolibc/picolibc/blob/main/cmake/picolibc.cmake#L123) function solves the problem.

keith-packard, 2024-08-13:
Odd. When I run the sample cmake-clang-thumbv7m configuration, I get the right result:
```
    kind: "try_compile-v1"
    backtrace:
      - "cmake/picolibc.cmake:151 (try_compile)"
      - "CMakeLists.txt:198 (picolibc_flag)"
    directories:
      source: "/home/keithp/src/picolibc/builds/build-cmake-clang-thumbv7m/CMakeFiles/CMakeTmp"
      binary: "/home/keithp/src/picolibc/builds/build-cmake-clang-thumbv7m/CMakeFiles/CMakeTmp"
    cmakeVariables:
      CMAKE_C_FLAGS: ""
      CMAKE_C_FLAGS_DEBUG: "-g"
      CMAKE_EXE_LINKER_FLAGS: ""
    buildResult:
      variable: "_HAVE_BUILTIN_ISNANL"
      cached: true
      stdout: |
        Change Dir: '/home/keithp/src/picolibc/builds/build-cmake-clang-thumbv7m/CMakeFiles/CMakeTmp'
        
        Run Build Command(s): /bin/ninja -v cmTC_d8ad7
        [1/2] /usr/lib/ccache/clang   -Werror=missing-declarations -Werror=unknown-attributes -Werror=attributes -Werror=unsupported-floating-point-opt -Werror=ignored-optimization-argument -Werror=attribute-optimize -Wno-unused-command-line-argument -MD -MT CMakeFiles/cmTC_d8ad7.dir/have-builtin-isnanl.c.obj -MF CMakeFiles/cmTC_d8ad7.dir/have-builtin-isnanl.c.obj.d -o CMakeFiles/cmTC_d8ad7.dir/have-builtin-isnanl.c.obj -c /home/keithp/src/picolibc/cmake/have-builtin-isnanl.c
        FAILED: CMakeFiles/cmTC_d8ad7.dir/have-builtin-isnanl.c.obj 
        /usr/lib/ccache/clang   -Werror=missing-declarations -Werror=unknown-attributes -Werror=attributes -Werror=unsupported-floating-point-opt -Werror=ignored-optimization-argument -Werror=attribute-optimize -Wno-unused-command-line-argument -MD -MT CMakeFiles/cmTC_d8ad7.dir/have-builtin-isnanl.c.obj -MF CMakeFiles/cmTC_d8ad7.dir/have-builtin-isnanl.c.obj.d -o CMakeFiles/cmTC_d8ad7.dir/have-builtin-isnanl.c.obj -c /home/keithp/src/picolibc/cmake/have-builtin-isnanl.c
        warning: unknown warning option '-Werror=attribute-optimize'; did you mean '-Werror=attribute-warning'? [-Wunknown-warning-option]
        /home/keithp/src/picolibc/cmake/have-builtin-isnanl.c:2:29: error: use of unknown builtin '__builtin_isnanl' [-Wimplicit-function-declaration]
        int main(void) { return foo(__builtin_isnanl((long double)42)); }
                                    ^
        1 warning and 1 error generated.
        ninja: build stopped: subcommand failed.
        
      exitCode: 1
```
I'm not sure why it works though; the command line shown doesn't appear to contain the right flag.

niedzwiecki-dawid, 2024-08-14:
Yes, I've also seen in Zephyr that it works. We have our internally built toolchain, there can be some differences in default flags/behavior, not sure? I didn't go into more details, because despite the result I think  `-Werror=implicit-function-declaration` is necessary here, what do you think?

keith-packard, 2024-08-14:
> Yes, I've also seen in Zephyr that it works. We have our internally built toolchain, there can be some differences in default flags/behavior, not sure? I didn't go into more details, because despite the result I think `-Werror=implicit-function-declaration` is necessary here, what do you think?

It does seem like a good addition, yes. If you'll submit a PR that you've tested locally, I'd be happy to merge it.

closed 2024-08-22 by keith-packard