picolibc/picolibc #814: Build race when building tests

arichardson, 2024-08-26:
I updated my picolibc checkout today and got the following error:
`error: no such file or directory: '$HOME/build/picolibc-arm32-build/newlib/libc.a'` when building the tests. It sounds like there is a missing dependency here?

Retrying a few times eventually allows the build to succeed.

keith-packard, 2024-08-30:
Yeah, I was noticing similar issues. Bisecting to figure out where it broke is probably indicated. I'm flying rockets for the next week and won't be able to get to figuring this out until I'm back home though.

keith-packard, 2024-09-08:
Ok, I've got a patch merged which should help. Please see if #824 had any effect for you. I probably should have tied this issue to that PR, but I wasn't sure it was the same issue.

arichardson, 2024-09-10:
The build now works for me, so I imagine https://github.com/picolibc/picolibc/pull/824 fixed the issue. Thanks!

closed 2024-09-10 by arichardson