picolibc/picolibc #815: duplicate symbol: __aeabi_memcpy*

arichardson, 2024-08-26:
```
ld.lld: error: duplicate symbol: __aeabi_memcpy
>>> defined at aeabi_memcpy.S:21 ($HOME/llvm-project/compiler-rt/lib/builtins/arm/aeabi_memcpy.S:21)
>>>            aeabi_memcpy.S.obj:(__aeabi_memcpy) in archive $HOME/build/picolibc-arm32-build/local-libgcc/libgcc.a
>>> defined at aeabi_memcpy-armv7a.S:50 (../../picolibc/newlib/libc/machine/arm/aeabi_memcpy-armv7a.S:50)
>>>            memcpy.S.o:(.text+0x0) in archive $HOME/build/picolibc-arm32-build/newlib/libc.a
```

I can work around this by marking one of them as weak but I'm not sure that's an appropriate solution. I am not sure whether libc or libgcc/compiler-rt is supposed to provide this symbol.

keith-packard, 2024-08-30:
libgcc doesn't define these symbols. Probably because libc generally has "better" implementations most of the time which can be shared with the standard libc functions. I'm a bit surprised you're getting duplicates though -- that would indicate that there is an undefined symbol referenced by the application which is not in picolibc but is in aeabi_memcpy.S. It would be good to know what other symbols are defined by aeabi_memcpy.S which picolibc doesn't define.

arichardson, 2024-08-30:
The problem is that I am using compiler-rt builtins symlinked as libgcc.a and compiler-rt builds [unconditionally include the aebi symbols](https://github.com/llvm/llvm-project/blob/main/compiler-rt/lib/builtins/CMakeLists.txt#L448C5-L448C21). So having this symbol provided by picolibc creates the duplicate symbol error.

I can definitely work around this locally by removing the symbol from compilert-builtins.a locally but I filed this issue since I would image others could run into it too (e.g. the people from https://github.com/ARM-software/LLVM-embedded-toolchain-for-Arm ?).

keith-packard, 2024-08-31:
A duplicate symbol from two libraries (A and B) can only occur if you have a file (file a) from library A which defines a symbol required by the application (symbol a) and a file (file b) from library B which defines a different symbol (symbol b) required by the application and then a symbol defined by file a (symbol c) which is the same as a symbol defined by file b. So, my question is what is symbol a, file a, symbol b and file b. If we know that, we might be able to define symbol b in picolibc so that file b from library B need not be included so that we don't end up with symbol c being defined twice.

closed 2024-09-06 by keith-packard