picolibc/picolibc #819: riscv: FAILED: test/test-cplusplus.p/test-cplusplus.cpp.o -> cc1plus: error: '-fcf-protection=full' is not supported for this target

iam-TJ, 2024-08-30:
Building with commit HEAD at `1584f3aba` on/for Debian unstable (2024-08-30) results in:
```
[33934/42810] riscv64-unknown-elf-g++ -nostdlib -Itest/test-cplusplus.p -Itest -I../../../test -Inewlib/libm/common -I../../../newlib/libm/common -Inewlib/libc/machine/riscv
-I../../../newlib/libc/machine/riscv -Inewlib/libc/tinystdio -I../../../newlib/libc/tinystdio -I. -I../../.. -Inewlib/libc/include -I../../../newlib/libc/include -Isemihost
-I../../../semihost -fdiagnostics-color=always -D_GLIBCXX_ASSERTIONS=1 -D_FILE_OFFSET_BITS=64 -Wall -Winvalid-pch -Wextra -Os -g -g -O2 -ffile-prefix-map=/<<PKGBUILDDIR>>=.
-fstack-protector-strong -fstack-clash-protection -Wformat -Werror=format-security -fcf-protection -mcmodel=medany -nostdlib -D_FORTIFY_SOURCE=3 -MD -MQ test/test-cplusplus
.p/test-cplusplus.cpp.o -MF test/test-cplusplus.p/test-cplusplus.cpp.o.d -o test/test-cplusplus.p/test-cplusplus.cpp.o -c ../../../test/test-cplusplus.cpp
FAILED: test/test-cplusplus.p/test-cplusplus.cpp.o
riscv64-unknown-elf-g++ -nostdlib -Itest/test-cplusplus.p -Itest -I../../../test -Inewlib/libm/common -I../../../newlib/libm/common -Inewlib/libc/machine/riscv -I../../../newlib/libc/machine/riscv -Inewlib/libc/tinystdio -I../../../newlib/libc/tinystdio -I. -I../../.. -Inewlib/libc/include -I../../../newlib/libc/include -Isemihost -I../../../semihost -fdiagnostics-color=always -D_GLIBCXX_ASSERTIONS=1 -D_FILE_OFFSET_BITS=64 -Wall -Winvalid-pch -Wextra -Os -g -g -O2 -ffile-prefix-map=/<<PKGBUILDDIR>>=. -fstack-protector-strong -fstack-clash-protection -Wformat -Werror=format-security -fcf-protection -mcmodel=medany -nostdlib -D_FORTIFY_SOURCE=3 -MD -MQ test/test-cplusplus.p/test-cplusplus.cpp.o -MF test/test-cplusplus.p/test-cplusplus.cpp.o.d -o test/test-cplusplus.p/test-cplusplus.cpp.o -c ../../../test/test-cplusplus.cpp
cc1plus: error: '-fcf-protection=full' is not supported for this target
...
ninja: build stopped: subcommand failed.
```

keith-packard, 2024-08-31:
Can you paste in your complete meson run so I can try to reproduce this? Using `scripts/do-riscv-configure` works fine for me on a fairly recent unstable machine (maybe a couple of days since my last apt get dist-upgrade?)

iam-TJ, 2024-08-31:
The issue is due to the default g++ compiler flags set in Debian Unstable which are valid for the host architecture (amd64) but not for riscv [0] and meson is using those. I'll attach a complete build log later - right now I'm in the middle of figuring out another test failure.

I figured out part of Debian bug #1077452 [2] which was originally thought to be due to Qemu but turns out Qemu 9 started doing 'correct' behaviour for ARM [1] and it caused FAILed tests due to a problem in ARM picocrt tests (MMU not enabled) that commit  5252f4dd236e4eb1a fixes.

I'm now left with a lot of (30+) TIMEOUT failures in the aarch64 tests with HEAD at `1584f3aba 2024-08-29 19:26:46 +0700 E Alexey Lapshin fix FD_ZERO macro typo` that I'm trying to get a handle on - they seem to be semi-random and I'll post another bug report about those when I've collected sufficient actionable information.

It's worth noting that I'm reproducing the Debian buildd sbuild invocation (since the aim is to fix the Debian build failure(s) [2] ) using `sbuild -n -A -s --force-orig-source --apt-update -d unstable -v --no-run-lintian`

[0] https://gcc.gnu.org/onlinedocs/gcc/Instrumentation-Options.html#index-fcf-protection
[0] https://gitlab.com/qemu-project/qemu/-/issues/2542#note_2081231016
[2] https://bugs.debian.org/cgi-bin/bugreport.cgi?bug=1077452

iam-TJ, 2024-08-31:
[sbuild.origin-main.test_cplusplus.log.tar.gz](https://github.com/user-attachments/files/16826040/sbuild.origin-main.test_cplusplus.log.tar.gz)
sbuild log (44MiB uncompressed) - search for:
```
FAILED: test/test-cplusplus.p/test-cplusplus.cpp.o
riscv64-unknown-elf-g++ -nostdlib -Itest/test-cplusplus.p -Itest -I../../../test -Inewlib/libm/common -I../../../newlib/libm/common -Inewlib/libc/machine/riscv -I../../../newlib/libc/machine/riscv -Inewlib/libc/tinystdio -I../../../newlib/libc/tinystdio -I. -I../../.. -Inewlib/libc/include -I../../../newlib/libc/include -Isemihost -I../../../semihost-fdiagnostics-color=always -D_GLIBCXX_ASSERTIONS=1 -D_FILE_OFFSET_BITS=64 -Wall -Winvalid-pch -Wextra -Os -g -g -O2 -ffile-prefix-map=/<<PKGBUILDDIR>>=. -fstack-protector-strong -fstack-clash-protection -Wformat -Werror=format-security -fcf-protection -mcmodel=medany -nostdlib -D_FORTIFY_SOURCE=3 -MD -MQ test/test-cplusplus.p/test-cplusplus.cpp.o -MF test/test-cplusplus.p/test-cplusplus.cpp.o.d -o test/test-cplusplus.p/test-cplusplus.cpp.o -c ../../../test/test-cplusplus.cpp
cc1plus: error: '-fcf-protection=full' is not supported for this target
```

keith-packard, 2024-09-05:
I'm working to get upstream to a point where I can start doing a new debian version to fix the test failures; I was hoping to get that done quickly but there are a bunch of new contributors fixing bugs and I'd love to get those merged first. Maybe it's time to do time-based releases of picolibc...

closed 2026-11-10 by keith-packard