picolibc/picolibc #820: aarch64 tests: many tests TIMEOUT

iam-TJ, 2024-08-31:
As mentioned in #819 Debian builds seem to be hitting  test `TIMEOUT`  failures. This is building HEAD @ `1584f3aba 2024-08-29 19:26:46 +0700 E Alexey Lapshin fix FD_ZERO macro typo` using the Debian package and buildd sbuild invocation: `sbuild -n -A -s --force-orig-source --apt-update -d unstable -v --no-run-lintian`
Generally I end up killing the job rather than waiting since it can take hours. As well as allowing parallel tests I used `MESON_TESTTHREADS=1` in one run (that I interrupted after about 5 hours!) so it doesn't seem to be a builder-related race of any sort.
```
$ for l in ../sbuild.*.log; do echo "$l FAIL=$(grep -nE ' FAIL ' $l | wc -l) TIMEOUT=$(grep -nE ' TIMEOUT ' $l | wc -l)"; done
../sbuild.origin-main.log FAIL=0 TIMEOUT=36
../sbuild.origin-main.test_cplusplus.log FAIL=0 TIMEOUT=36
../sbuild.origin-main.TESTTHREADS=1.log FAIL=0 TIMEOUT=32
```
Looking at the detail from the single-thread run where the ordering is serial:
```
../sbuild.origin-main.TESTTHREADS=1.log TIMEOUT
204773: 89/193 picolibc / nulprintf                               TIMEOUT         300.11s   killed by signal 15 SIGTERM
204781: 95/193 picolibc / tstring                                 TIMEOUT         300.11s   killed by signal 15 SIGTERM
204784: 96/193 picolibc / tzset                                   TIMEOUT         300.11s   killed by signal 15 SIGTERM
204787: 97/193 picolibc / asctime                                 TIMEOUT         300.06s   killed by signal 15 SIGTERM
204791: 99/193 picolibc / twctrans                                TIMEOUT         300.11s   killed by signal 15 SIGTERM
204794:100/193 picolibc / math_test                               TIMEOUT         300.11s   killed by signal 15 SIGTERM
204797:101/193 picolibc / printf_scanf                            TIMEOUT         300.01s   killed by signal 15 SIGTERM
204800:102/193 picolibc / printff_scanff                          TIMEOUT         300.02s   killed by signal 15 SIGTERM
204803:103/193 picolibc / printfl_scanfl                          TIMEOUT         300.11s   killed by signal 15 SIGTERM
204806:104/193 picolibc / printfi_scanfi                          TIMEOUT         300.11s   killed by signal 15 SIGTERM
204809:105/193 picolibc / printfm_scanfm                          TIMEOUT         300.11s   killed by signal 15 SIGTERM
204812:106/193 picolibc / printf-tests                            TIMEOUT         300.11s   killed by signal 15 SIGTERM
204815:107/193 picolibc / printff-tests                           TIMEOUT         300.11s   killed by signal 15 SIGTERM
204818:108/193 picolibc / printfi-tests                           TIMEOUT         300.11s   killed by signal 15 SIGTERM
204821:109/193 picolibc / printfm-tests                           TIMEOUT         300.10s   killed by signal 15 SIGTERM
204824:110/193 picolibc / time-sprintf                            TIMEOUT         300.11s   killed by signal 15 SIGTERM
204827:111/193 picolibc / try-ilp32                               TIMEOUT         300.10s   killed by signal 15 SIGTERM
204839:121/193 picolibc / test-long-longl                         TIMEOUT         600.11s   killed by signal 15 SIGTERM
204842:122/193 picolibc / test-long-long                          TIMEOUT         600.11s   killed by signal 15 SIGTERM
204852:130/193 picolibc / timegm                                  TIMEOUT         600.11s   killed by signal 15 SIGTERM
204862:138/193 picolibc / t_fmemopen                              TIMEOUT         600.11s   killed by signal 15 SIGTERM
204865:139/193 picolibc / test-vfscanf-percent-a                  TIMEOUT         600.10s   killed by signal 15 SIGTERM
204869:141/193 picolibc / posix-io                                TIMEOUT         600.11s   killed by signal 15 SIGTERM
204872:142/193 picolibc / test-fopen                              TIMEOUT         600.11s   killed by signal 15 SIGTERM
204875:143/193 picolibc / test-mktemp                             TIMEOUT         600.11s   killed by signal 15 SIGTERM
204879:145/193 picolibc / test-fread-fwrite                       TIMEOUT         600.11s   killed by signal 15 SIGTERM
204882:146/193 picolibc / test-ungetc-ftell                       TIMEOUT         600.02s   killed by signal 15 SIGTERM
204886:148/193 picolibc / test-fgets-eof                          TIMEOUT         600.10s   killed by signal 15 SIGTERM
204896:156/193 picolibc / malloc_stress                           TIMEOUT         600.11s   killed by signal 15 SIGTERM
204900:158/193 picolibc / test-funopen                            TIMEOUT         600.11s   killed by signal 15 SIGTERM
204903:159/193 picolibc / time-tests                              TIMEOUT         600.11s   killed by signal 15 SIGTERM
204907:161/193 picolibc / stack-smash                             TIMEOUT         600.11s   killed by signal 15 SIGTERM
```
When it happens the hung test processes look like this and are usually consuming 100% CPU:
```
S 166534   3568836 3568834 99  80   0 60264 386286 do_sys 17:24 ?       00:47:27 qemu-system-aarch64 -chardev stdio,mux=on,id=stdio0 -semihosting-config enable=on,chardev=stdio0,arg=program-name -monitor none -serial none -M virt -cpu cortex-a57 -nographic -kernel /build/picolibc-yoPLP0/picolibc-1.8.6/debian/build/aarch64-linux-gnu/test/libc-testsuite/snprintf -nic none
```
and Meson will report something like (excuse the different tests; I didn't capture the process info for `string` ):
```
190/193 picolibc / string                                  TIMEOUT         300.13s   killed by signal 15 SIGTERM
>>> PATH=/<<PKGBUILDDIR>>/scripts:/usr/local/sbin:/usr/local/bin:/usr/sbin:/usr/bin:/sbin:/bin:/usr/games MSAN_OPTIONS=halt_on_error=1:abort_on_error=1:print_summary=1:print_st
acktrace=1 PICOLIBC_TEST=1 ASAN_OPTIONS=halt_on_error=1:abort_on_error=1:print_summary=1 MESON_EXE_WRAPPER='sh -c '"'"'test -z "$PICOLIBC_TEST" || run-aarch64 "$@"'"'"' run-aar
ch64' UBSAN_OPTIONS=halt_on_error=1:abort_on_error=1:print_summary=1:print_stacktrace=1 MESON_TEST_ITERATION=1 MALLOC_PERTURB_=243 sh -c 'test -z "$PICOLIBC_TEST" || run-aarch6
4 "$@"' run-aarch64 /<<PKGBUILDDIR>>/debian/build/aarch64-linux-gnu/test/libc-testsuite/string
```
When `test-funopen` failed and the sbuild was ongoing so I could access the binaries and do a cross-arch remote `gdb-multiarch` session.

Terminal A:
```
$ sudo qemu-system-aarch64 -gdb tcp::9999 -S -chardev stdio,mux=on,id=stdio0 -semihosting-config enable=on,chardev=stdio0,arg=program-name -monitor none -serial none -M virt -cpu cortex-a57 -nographic -kernel /srv/NAS/Sunny/SourceCode/builds/tmp.sbuild.TvHIcX5w/build/picolibc-yoPLP0/picolibc-1.8.6/debian/build/aarch64-linux-gnu/test/test-funopen -nic none
```
Terminal B:
```
picolibc/picolibc-1.8.6$
 base="/srv/NAS/Sunny/SourceCode/builds";
 sbuild="$( sudo sh -c "cd $base; find . -maxdepth 1 -type d -name 'tmp.sbuild.*' | tail -n 1" )";
 picolibc="$( sudo sh -c "cd $base/$sbuild/build; find . -maxdepth 1 -type d -name 'picolibc-*'" )";
 sudo gdb-multiarch -ex "directory $base/$sbuild/build/$picolibc/picolibc-1.8.6/debian/build/aarch64-linux-gnu" -ex "file $base/$sbuild/build/$picolibc/picolibc-1.8.6/debian/build/aarch64-linux-gnu/test/test-funopen" -ex "set sysroot $base/$sbuild" -ex "target remote localhost:9999" -ex "break main"
```
[... 110 lines cut ...]
Cannot find bounds of current function
(gdb) n
Cannot find bounds of current function
(gdb) finish
"finish" not meaningful in the outermost frame.
(gdb) bt
#0  0x0000000000000200 in ?? ()
(gdb) c
Continuing.
^C
Program received signal SIGINT, Interrupt.
0x0000000000000200 in ?? ()
(gdb) bt
#0  0x0000000000000200 in ?? ()
(gdb)
```
In an earlier similar debug session (and I cannot now be sure it was for aarch64 or arm since I was debugging Debian's 1.8.6-2 package source that was hitting these TIMEOUTs, not origin/main) I traced the lock tests and that seemed to spin either the `_start` or `_exit` related code - I'm wondering if the debug session above is affected similarly since there is a mention of `__retarget_lock_acquire () at ../../../newlib/libc/misc/lock.c:111`

My build host is Debian Bookworm amd64 using Qemu from bookworm-backports (same version as in Unstable):
```
$ qemu-system-aarch64 --version
QEMU emulator version 9.0.2 (Debian 1:9.0.2+ds-1~bpo12+1)
Copyright (c) 2003-2024 Fabrice Bellard and the QEMU Project developers
```
I'm currently letting a build run to completion (it is intended to capture the risvc `test_cplusplus` failure log) but it hasn't completed the aarch64 tests yet due to all the TIMEOUTs. The logs are large (~240MiB) so I may have to host it and provide a link.

iam-TJ, 2024-08-31:
The aarch64 `picolibc / snprintf` debug trace:
```
_start () at ../../../picocrt/machine/aarch64/crt0.c:73
73              __asm__("adrp x1, __stack");
Breakpoint 1 at 0x40000038: file ../../../test/libc-testsuite/testcase.h, line 52.
(gdb) n
74              __asm__("add  x1, x1, :lo12:__stack");
(gdb) n
75              __asm__("mov sp, x1");
(gdb) n

Breakpoint 1, main () at ../../../test/libc-testsuite/testcase.h:52
52          RUN_TEST(TEST_NAME);
(gdb) s
0x000000004000016c in test_snprintf () at ../../../test/libc-testsuite/snprintf.c:179
179             TEST_S(b, "123", "incorrect output");
(gdb) s
177             strcpy(b, "xxxxxxxx");
(gdb) n
178             TEST(i, snprintf(b, 4, "%ld", 123456l), 6, "length returned %d != %d");
(gdb) s
snprintf (s=s@entry=0x405ff7a8 "xxxxxxxx", n=n@entry=4, fmt=fmt@entry=0x40002629 "%ld") at ../../../newlib/libc/tinystdio/snprintf.c:46
46              if ((int) n < 0)
(gdb) n
49              struct __file_str f = FDEV_SETUP_STRING_WRITE(s, n ? n - 1 : 0);
(gdb) s
0x0000000000000200 in ?? ()
(gdb) s
Cannot find bounds of current function
(gdb) c
Continuing.
^C
Program received signal SIGINT, Interrupt.
0x0000000000000200 in ?? ()
(gdb) bt
#0  0x0000000000000200 in ?? ()
(gdb)
```

keith-packard, 2024-10-02:
This happened when qemu was updated and started respecting the memory configuration bits on arm/aarch64 targets with an MMU. This should have been fixed by #799

closed 2024-10-02 by keith-packard