picolibc/picolibc #845: How to tweak entrance when using picocrt

yf13, 2024-09-27:
Hi, I am using `picolibc-riscv64-unknown-elf` on Ubuntu 22.04 for QEMU riscv targets and so far so good with single core devices. My use is based on the `-specs=picolibc.specs` and `--crt0=hosted`. So far the experience is so good!

Now I want use picolibc with the dual core riscv target. Guess I need add a new `mystart.S` to my project, where the hartid is used to dispatch cores to different routes: e.g. core 0 to continue with Picocrt and core 1 to a waiting loop.

I am wondering how to make it work with `picolibc.specs`?

  - How can I make `mystart.S` as the entry of my program?
  - How can I refer to the Picolibc entrance in `mystart.S`?

keith-packard, 2024-09-27:
Probably the best plan is to take the sample startup code from picocrt and copy that into your project, then modify that to support multiple cores. In my experience, all of the cores start at the same address. One core initializes the system while the others spin waiting for a signal, then the other cores jump to secondary core startup code. I'd take a look at how a more complete RTOS like Zephyr does initialization on multi-core RISC-V targets.

yf13, 2024-09-27:
@keith-packard, very appreciate your quick response.

Yes, my starting point is a copy of `picocrt/machine/riscv/crt0.S` to my project as `start.S`. However, I don't know the option to use with `--crt0=` so that to make it work with `-specs=picolibc.spec --oslib=semihost` options. It seems that passing absolute path to `--crt0=` doesn't work yet.

keith-packard, 2024-09-28:
> is there a cleaner way to skip using any pre-built `crt0`?

Hrm. Not really. We should probably fix `picolibc.specs` so that you can provide something like --crt0=none and have it not use any of the provided ones. As you've already discovered, that can be done trivially by creating an empty object file. Fixing picocrt to build and install one like that shouldn't take long. Let me cook up a patch.

keith-packard, 2024-10-02:
Somehow github didn't catch that the PR was merged.

closed 2024-10-02 by keith-packard