picolibc/picolibc #849: Build failure with clang for x86

PwnVerse, 2024-10-01:
Picolibc build fails with clang-10 for x86 both 32 and 64 bit. Here is the complete log of what I tried, please let me know if I'm missing something trivial in any required configuration - 

```
picolibc/clang_build$ ../scripts/do-clang-x86-configure
The Meson build system
Version: 1.5.2
Source dir: /workdisk/disk/picolibc
Build dir: /workdisk/disk/picolibc/clang_build
Build type: cross build
Project name: picolibc
Project version: 1.8.8
Cross compiler sanity tests disabled via the cross file.
C compiler for the host machine: clang -target i686-none-elf -march=core2 -mfpmath=sse -msse2 -fno-pic -fno-PIE -static -nostdlib (clang 10.0.1 "clang version 10.0.1 ")
C linker for the host machine: clang -target i686-none-elf -march=core2 -mfpmath=sse -msse2 -fno-pic -fno-PIE -static -nostdlib ld.lld 10.0.1
C compiler for the build machine: ccache cc (gcc 11.4.0 "cc (Ubuntu 11.4.0-1ubuntu1~22.04) 11.4.0")
C linker for the build machine: cc ld.bfd 2.38
Build machine cpu family: x86_64
Build machine cpu: x86_64
Host machine cpu family: x86
Host machine cpu: x86
Target machine cpu family: x86
Target machine cpu: x86
Compiler for C supports arguments -nostdlib: YES
Checking if "long double check" compiles: YES
C++ compiler for the build machine: ccache c++ (gcc 11.4.0 "c++ (Ubuntu 11.4.0-1ubuntu1~22.04) 11.4.0")
C++ linker for the build machine: c++ ld.bfd 2.38
C++ compiler for the host machine: clang -target i686-none-elf -march=core2 -mfpmath=sse -msse2 -fno-pic -fno-PIE -static -nostdlib (clang 10.0.1 "clang version 10.0.1 ")
C++ linker for the host machine: clang -target i686-none-elf -march=core2 -mfpmath=sse -msse2 -fno-pic -fno-PIE -static -nostdlib ld.lld 10.0.1
Compiler for C++ supports arguments -fno-common: YES
Compiler for C++ supports arguments -frounding-math: YES
Compiler for C++ supports arguments -fsignaling-nans: NO
Compiler for C++ supports arguments -Wno-unsupported-floating-point-opt: NO
Compiler for C++ supports arguments -fno-builtin-copysignl: YES
Compiler for C supports arguments -fno-common: YES
Compiler for C supports arguments -frounding-math: YES
Compiler for C supports arguments -fsignaling-nans: NO
Compiler for C supports arguments -Wno-unsupported-floating-point-opt: NO
Compiler for C supports arguments -fno-builtin-copysignl: YES
Compiler for C supports arguments -fsanitize=bounds: YES
Compiler for C supports arguments -fsanitize-undefined-trap-on-error: YES
Compiler for C supports arguments -fno-sanitize-undefined-trap-on-error: YES
Compiler for C supports arguments -static-libubsan: NO
Program nm found: YES
Program scripts/duplicate-names found: YES (/workdisk/disk/picolibc/scripts/duplicate-names)
Program scripts/validate-cdefs found: YES (/workdisk/disk/picolibc/scripts/validate-cdefs)
Compiler for C supports link arguments -Wl,--defsym=start=0: NO
Compiler for C supports link arguments -Wl,--defsym=_start=0: YES
Compiler for C supports link arguments -Wl,-alias,main,testalias: NO
Compiler for C supports function attribute alias: YES
Compiler for C supports function attribute format: YES
Compiler for C supports function attribute weak: YES
Configuring picolibc.specs using configuration
Configuring picolibcpp.specs using configuration
Compiler for C supports arguments -Werror=vla: YES
Compiler for C supports arguments -Warray-bounds: YES
Compiler for C supports arguments -Werror=double-promotion: YES
Compiler for C supports arguments -Wno-missing-braces: YES
Compiler for C supports arguments -Wno-return-type: YES
Compiler for C supports arguments -Wno-unused-command-line-argument: YES
Compiler for C supports arguments -Wmissing-prototypes: YES
Compiler for C supports arguments -Wmissing-declarations: YES
Compiler for C supports arguments -Werror=implicit-fallthrough=5: NO
Compiler for C supports arguments -Werror=unreachable-code-fallthrough: NO
Compiler for C supports arguments -Werror=implicit-function-declaration: YES
Compiler for C supports arguments -Wold-style-definition: YES
Compiler for C supports arguments -Wno-implicit-int: YES
Compiler for C supports arguments -fno-stack-protector: YES
Compiler for C++ supports arguments -Werror=vla: YES
Compiler for C++ supports arguments -Warray-bounds: YES
Compiler for C++ supports arguments -Werror=double-promotion: YES
[... 125 lines cut ...]

Found ninja-1.11.1.git.kitware.jobserver-1 at /home/ritvik/.local/bin/ninja

picolibc/clang_build$ ninja
[14/1338] Compiling C object picocrt/crt0-minimal.o.p/machine_x86_crt0.S.o
FAILED: picocrt/crt0-minimal.o.p/machine_x86_crt0.S.o
clang -target i686-none-elf -march=core2 -mfpmath=sse -msse2 -fno-pic -fno-PIE -static -nostdlib -Ipicocrt/crt0-minimal.o.p -Ipicocrt -I../picocrt -Inewlib/libc/tinystdio -I../newlib/libc/tinystdio -I. -I.. -Inewlib/libc/include -I../newlib/libc/include -fdiagnostics-color=always -D_FILE_OFFSET_BITS=64 -Wall -Winvalid-pch -Wextra -std=c18 -Oz -g -ffunction-sections -fno-builtin -ffreestanding -DCONSTRUCTORS=0 -MD -MQ picocrt/crt0-minimal.o.p/machine_x86_crt0.S.o -MF picocrt/crt0-minimal.o.p/machine_x86_crt0.S.o.d -o picocrt/crt0-minimal.o.p/machine_x86_crt0.S.o -c ../picocrt/machine/x86/crt0.S
../picocrt/machine/x86/crt0-32.S:48:11: error: unexpected token in argument list
 cs lgdtl gdt_desc - _start + 0x10
          ^
[22/1338] Compiling C object picocrt/crt0.o.p/machine_x86_crt0.S.o
FAILED: picocrt/crt0.o.p/machine_x86_crt0.S.o
clang -target i686-none-elf -march=core2 -mfpmath=sse -msse2 -fno-pic -fno-PIE -static -nostdlib -Ipicocrt/crt0.o.p -Ipicocrt -I../picocrt -Inewlib/libc/tinystdio -I../newlib/libc/tinystdio -I. -I.. -Inewlib/libc/include -I../newlib/libc/include -fdiagnostics-color=always -D_FILE_OFFSET_BITS=64 -Wall -Winvalid-pch -Wextra -std=c18 -Oz -g -ffunction-sections -fno-builtin -ffreestanding -MD -MQ picocrt/crt0.o.p/machine_x86_crt0.S.o -MF picocrt/crt0.o.p/machine_x86_crt0.S.o.d -o picocrt/crt0.o.p/machine_x86_crt0.S.o -c ../picocrt/machine/x86/crt0.S
../picocrt/machine/x86/crt0-32.S:48:11: error: unexpected token in argument list
 cs lgdtl gdt_desc - _start + 0x10
          ^
[23/1338] Compiling C object picocrt/crt0-hosted.o.p/machine_x86_crt0.S.o
FAILED: picocrt/crt0-hosted.o.p/machine_x86_crt0.S.o
clang -target i686-none-elf -march=core2 -mfpmath=sse -msse2 -fno-pic -fno-PIE -static -nostdlib -Ipicocrt/crt0-hosted.o.p -Ipicocrt -I../picocrt -Inewlib/libc/tinystdio -I../newlib/libc/tinystdio -I. -I.. -Inewlib/libc/include -I../newlib/libc/include -fdiagnostics-color=always -D_FILE_OFFSET_BITS=64 -Wall -Winvalid-pch -Wextra -std=c18 -Oz -g -ffunction-sections -fno-builtin -ffreestanding -DCRT0_EXIT -MD -MQ picocrt/crt0-hosted.o.p/machine_x86_crt0.S.o -MF picocrt/crt0-hosted.o.p/machine_x86_crt0.S.o.d -o picocrt/crt0-hosted.o.p/machine_x86_crt0.S.o -c ../picocrt/machine/x86/crt0.S
../picocrt/machine/x86/crt0-32.S:48:11: error: unexpected token in argument list
 cs lgdtl gdt_desc - _start + 0x10
          ^
[39/1338] Compiling C object newlib/libc/stdlib/libstdlib_malloc.a.p/nano-realloc.c.o
ninja: build stopped: subcommand failed.
```

keith-packard, 2024-10-01:
You're just using a compiler that doesn't support the full x86 assembly syntax. This patch worked for me using clang-11, maybe it will work for you?
```
diff --git a/picocrt/machine/x86/crt0-32.S b/picocrt/machine/x86/crt0-32.S
index 78f9248a7..f5b584bce 100644
--- a/picocrt/machine/x86/crt0-32.S
+++ b/picocrt/machine/x86/crt0-32.S
@@ -45,7 +45,8 @@
        .type   start, @function
 _start:
        .code16
-       cs lgdtl gdt_desc - _start + 0x10
+       .byte 0x2e                      # CS override prefix
+       lgdtl (gdt_desc - _start + 0x10)
        mov $1, %eax
        mov %eax, %cr0
        ljmpl $0x10,$_start32
```

keith-packard, 2024-10-01:
oh, looks like clang-11 didn't really support prefixes; they appear to have been treated more like regular instructions. Placing it on a separate line appears to work fine, and also works with the GNU assembler.
```
diff --git a/picocrt/machine/x86/crt0-32.S b/picocrt/machine/x86/crt0-32.S
index 78f9248a7..793471481 100644
--- a/picocrt/machine/x86/crt0-32.S
+++ b/picocrt/machine/x86/crt0-32.S
@@ -45,7 +45,8 @@
        .type   start, @function
 _start:
        .code16
-       cs lgdtl gdt_desc - _start + 0x10
+       cs
+       lgdtl gdt_desc - _start + 0x10
        mov $1, %eax
        mov %eax, %cr0
        ljmpl $0x10,$_start32
```
That's a lot more reasonable; I'll post a PR and make sure it doesn't break anything with CI.

PR #851 (2024-10-02): picocrt: Adapt x86 crt0 code for old clang assembler
Before clang-12, the built-in x86 assembler seems to treat prefixes as separate instructions. This meant that it wouldn't allow them on the same source line as the rest of the instruction. Split out the 'cs' prefix used in the x86 crt0 code to a separate line so that old clang versions can also assemble the code.

Closes: #849
closed 2024-10-02 by keith-packard