picolibc/picolibc #880: Compiling with `mlong-double-128`

zyedidia, 2024-11-08:
I'm trying to compile picolibc with `-mlong-double-128` so that the long double implementation will use SSE instructions rather than x87 (I am also passing `-mno-80387`). However, when I try this I get errors like the following

```
../picolibc/newlib/libm/ld/ld80/s_erfl.c: In function ‘erfl’:
../picolibc/newlib/libm/ld/ld80/s_erfl.c:253:3: error: implicit declaration of function ‘GET_LDOUBLE_WORDS’; did you mean ‘GET_LDOUBLE_WORDS64’? [-Wimplicit-function-declaration]
  253 |   GET_LDOUBLE_WORDS (se, i0, i1, x);
      |   ^~~~~~~~~~~~~~~~~
      |   GET_LDOUBLE_WORDS64
../picolibc/newlib/libm/ld/ld80/s_erfl.c:317:3: error: implicit declaration of function ‘SET_LDOUBLE_WORDS’; did you mean ‘SET_LDOUBLE_WORDS64’? [-Wimplicit-function-declaration]
  317 |   SET_LDOUBLE_WORDS (z, i, i0, i1);
      |   ^~~~~~~~~~~~~~~~~
      |   SET_LDOUBLE_WORDS64
```

It looks like for some reason in `s_erfl.c` the constant `LDBL_MANT_DIG` is set to 64 rather than 113, even though `__LDBL_MANT_DIG__` is properly set by the compiler to 113. I've dug around a bit and haven't been able to figure out where `LDBL_MANT_DIG` is defined, and why it might be 64 even though `__LDBL_MANT_DIG__` is 113. Any help would be appreciated, thank you!

keith-packard, 2024-11-11:
`LDBL_MANT_DIG` comes from the C compiler's `float.h` header. I can't see how it would ever fail to match `__LDBL_MANT_DIG__` though -- float.h from both clang and gcc have the same line:
```
#define LDBL_MANT_DIG __LDBL_MANT_DIG__
```

keith-packard, 2024-11-12:
Ok, using GCC, this seems to "work", except that I don't have any libgcc.a implementation of those functions.
```
c = ['x86_64-zephyr-elf-gcc', '-mno-80387', '-mlong-double-128', '-fno-pic', '-fno-PIE', '-static', '-nostdlib']
```
When building stuff, I get a lot of lines like this:
```
../../newlib/libm/test/convert.c:117: undefined reference to `__lttf2'
```
That means it's trying to use the 128-bit long-double code, but failing to find the software fallbacks. I'm reasonably sure there is no 128-bit hardware float support for x86, so you'd need a soft float implementation.

zyedidia, 2024-11-13:
I see thanks, I think this issue was caused by a misconfiguration in my compiler toolchain, not a problem with picolibc.

closed 2024-11-13 by zyedidia