picolibc/picolibc #900: Issues with upstream tools

FlyGoat, 2024-12-28:
Following #898 I'm now attempting to migrate CI tools to latest upstream release (GCC 14.2, QEMU 9.2.0).

I've got many successes so fat, including msp430 toolchains (with GDB sim built from picolibc fork), avr toolchain, m68k toolchain, arc QEMU.

However, I've got issues on following components:

## GCC sh4

Seems to be an FPU issue, I think underflow/overflow are not signally properly.

Full log: https://github.com/FlyGoat/picolibc/actions/runs/12526510671/job/34939275960

```
2024-12-28T13:09:16.1053156Z Summary of Failures:
2024-12-28T13:09:16.1053458Z
2024-12-28T13:09:16.1053809Z  311/1299 picolibc / math_test_m2e                            FAIL               1.19s   exit status 1
2024-12-28T13:09:16.1054606Z  310/1299 picolibc / math_test_m2                             FAIL               1.37s   exit status 1
2024-12-28T13:09:16.1055568Z  308/1299 picolibc / math_test                                FAIL               1.61s   exit status 1
2024-12-28T13:09:16.1056522Z  312/1299 picolibc / math_test_m4                             FAIL               0.42s   exit status 1
2024-12-28T13:09:16.1057293Z  309/1299 picolibc / math_test_mb                             FAIL               1.67s   exit status 1
2024-12-28T13:09:16.1058087Z  313/1299 picolibc / math_test_m4_single                      FAIL               0.49s   exit status 1
2024-12-28T13:09:16.1058858Z  316/1299 picolibc / math_test_mb_m4                          FAIL               0.39s   exit status 1
2024-12-28T13:09:16.1059664Z  317/1299 picolibc / math_test_mb_m4_single                   FAIL               0.40s   exit status 1
2024-12-28T13:09:16.1060479Z  315/1299 picolibc / math_test_mb_m2e                         FAIL               1.17s   exit status 1
2024-12-28T13:09:16.1061248Z  314/1299 picolibc / math_test_mb_m2                          FAIL               1.42s   exit status 1
2024-12-28T13:09:16.1062022Z  335/1299 picolibc / math_errhandling                         FAIL               0.21s   exit status 66
2024-12-28T13:09:16.1062852Z  395/1299 picolibc / math_errhandling_mb                      FAIL               0.20s   exit status 66
2024-12-28T13:09:16.1063697Z  455/1299 picolibc / math_errhandling_m2                      FAIL               0.20s   exit status 66
2024-12-28T13:09:16.1064731Z  515/1299 picolibc / math_errhandling_m2e                     FAIL               0.32s   (exit status 131 or signal 3 SIGQUIT)
2024-12-28T13:09:16.1065785Z  575/1299 picolibc / math_errhandling_m4                      FAIL               0.61s   exit status 21
2024-12-28T13:09:16.1066582Z  597/1299 picolibc / fenv_m4                                  FAIL               0.10s   exit status 2
2024-12-28T13:09:16.1067405Z  635/1299 picolibc / math_errhandling_m4_single               FAIL               0.70s   exit status 123
2024-12-28T13:09:16.1068393Z  695/1299 picolibc / math_errhandling_m4_single_only          FAIL               0.45s   (exit status 197 or signal 69 SIGinvalid)
2024-12-28T13:09:16.1069351Z  755/1299 picolibc / math_errhandling_mb_m2                   FAIL               0.21s   exit status 66
2024-12-28T13:09:16.1070270Z  815/1299 picolibc / math_errhandling_mb_m2e                  FAIL               0.29s   (exit status 131 or signal 3 SIGQUIT)
2024-12-28T13:09:16.1071165Z  875/1299 picolibc / math_errhandling_mb_m4                   FAIL               0.61s   exit status 21
2024-12-28T13:09:16.1071961Z  897/1299 picolibc / fenv_mb_m4                               FAIL               0.10s   exit status 2
2024-12-28T13:09:16.1072768Z  935/1299 picolibc / math_errhandling_mb_m4_single            FAIL               0.69s   exit status 123
2024-12-28T13:09:16.1073898Z  995/1299 picolibc / math_errhandling_mb_m4_single_only       FAIL               0.48s   (exit status 197 or signal 69 SIGinvalid)
2024-12-28T13:09:16.1074523Z
2024-12-28T13:09:16.1074649Z Ok:                 1043
2024-12-28T13:09:16.1075123Z Expected Fail:      24
2024-12-28T13:09:16.1075471Z Fail:               24
2024-12-28T13:09:16.1075801Z Unexpected Pass:    0
2024-12-28T13:09:16.1076148Z Skipped:            208
2024-12-28T13:09:16.1076476Z Timeout:            0
```

## GCC m68k

pow/powf accuracy problem.

Full log: https://github.com/FlyGoat/picolibc/actions/runs/12525821507/job/34937851282

```
2024-12-28T11:02:07.6598419Z pow:137, inaccurate answer: bit 0 (8000000000000000 0000000000000000) (-0 0)
2024-12-28T11:02:07.6598862Z testing powf
2024-12-28T11:02:07.6599351Z powf:137, inaccurate answer: bit 0 (80000000 is 00000000) (-0 is 0) (is double 0x00000000, 0x00000000)
```

## QEMU arm/m68k

Hang & timeout at random cases.

Might be a GitHub problem, havent look into it.

@keith-packard Do you mind to share if there is any patch you are carrying for those components, so I can intergrate them in toolchains or eventually upstream them?

Thanks!

keith-packard, 2024-12-28:
My sh-unknown-elf toolchain was built using crosstool-ng using the sample config. It looks like it was built from https://github.com/zephyrproject-rtos/crosstool-ng/commit/822807ab6b4806d1bc5614f71773d7bb4271403f

The m68k patches for libgcc are upstream and appear to also be in 14.1.0.

Most of the gcc patches which aren't upstream are included in crosstool-ng

I think all of my qemu changes for arm are upstream; the qemu-system-arm version was built from v9.0.0-2711-g93b799fafd. 

The m68k version was built from https://github.com/keith-packard/qemu/tree/m68k-mul2 -- looks like Richard Henderson has created a different version of the semihosting changes for upstream; I haven't tested to see if they work for picolibc.

FlyGoat, 2024-12-28:
Thanks for the info! I’ll try to fiddling around  versions.

FlyGoat, 2024-12-29:
For m68k issues:
pow/powf accuracy problem seems to be a GCC bug, downgrading to gcc 10.5 with libgcc patch applied do workaround it.

QEMU has a bug on handling semihosting, fixed and reported to upstream:
https://lore.kernel.org/qemu-devel/20241229-m68k-semihosting-v1-1-db131e2b5212@flygoat.com/

keith-packard, 2024-12-29:
gcc at basepoint/gcc-15 seems solid. gcc master is broken, but in a different way. I've got a patch for that which I'm sending to the author of the patch which broke it.

keith-packard, 2024-12-29:
argh. gcc upstream dropped part of my m68k soft float patch. That's causing all kinds of grief.

FlyGoat, 2024-12-29:
For SuperH, as per my digging into GCC, it seems like reversed NaN signalling bits and FP exceptions for soft-float is never handled properly in GCC. I don't understand how can CI work for now...

I just attempted to make a quick & dirty patch but it seems like there are still many test failures:

https://github.com/FlyGoat/gcc/commit/6970eccb94a2a867b4f80255dfd96b63c7c5e75c

Maybe you or @olegendo know better than me :-)

FlyGoat, 2024-12-29:
I think it's mostly libgcc not handling QNaN and SNaN in soft float properly after this patch:

For math_errhandling (softfloat):
```
Listing only the last 100 lines from a long log.
        bad value got sNaN expect qNaN
    log10_qnan                     = nan errno 0 (Success) except NONE
        bad value got sNaN expect qNaN
    log10_snan                     = nan errno 0 (Success) except NONE
        bad value got sNaN expect qNaN
    log1p_qnan                     = nan errno 0 (Success) except NONE
        bad value got sNaN expect qNaN
    log1p_snan                     = nan errno 0 (Success) except NONE
        bad value got sNaN expect qNaN
    log2_qnan                      = nan errno 0 (Success) except NONE
        bad value got sNaN expect qNaN
    log2_snan                      = nan errno 0 (Success) except NONE
        bad value got sNaN expect qNaN
    log2_neg                       = nan errno 0 (Success) except NONE
        bad value got sNaN expect qNaN
    log2_neginf                    = nan errno 0 (Success) except NONE
        bad value got sNaN expect qNaN
    logb_qnan                      = nan errno 0 (Success) except NONE
        bad value got sNaN expect qNaN
    logb_snan                      = nan errno 0 (Success) except NONE
        bad value got sNaN expect qNaN
    nearbyint_qnan                 = nan errno 0 (Success) except NONE
        bad value got sNaN expect qNaN
    nearbyint_snan                 = nan errno 0 (Success) except NONE
        bad value got sNaN expect qNaN
    pow_1_snan                     = nan errno 0 (Success) except NONE
        bad value got sNaN expect qNaN
    pow_snan_0                     = nan errno 0 (Success) except NONE
        bad value got sNaN expect qNaN
    pow_snan_neg0                  = nan errno 0 (Success) except NONE
        bad value got sNaN expect qNaN
    pow_2_qnan                     = nan errno 0 (Success) except NONE
        bad value got sNaN expect qNaN
    pow_2_snan                     = nan errno 0 (Success) except NONE
        bad value got sNaN expect qNaN
    pow10_qnan                     = nan errno 0 (Success) except NONE
        bad value got sNaN expect qNaN
    pow10_snan                     = nan errno 0 (Success) except NONE
        bad value got sNaN expect qNaN
    remainder_qnan_1               = nan errno 0 (Success) except NONE
        bad value got sNaN expect qNaN
    remainder_1_qnan               = nan errno 0 (Success) except NONE
        bad value got sNaN expect qNaN
    remainder_snan_1               = nan errno 0 (Success) except NONE
        bad value got sNaN expect qNaN
    remainder_1_snan               = nan errno 0 (Success) except NONE
        bad value got sNaN expect qNaN
    rint_qnan                      = nan errno 0 (Success) except NONE
        bad value got sNaN expect qNaN
    rint_snan                      = nan errno 0 (Success) except NONE
        bad value got sNaN expect qNaN
    scalb_qnan_1                   = nan errno 0 (Success) except NONE
        bad value got sNaN expect qNaN
    scalb_1_qnan                   = nan errno 0 (Success) except NONE
        bad value got sNaN expect qNaN
    scalb_snan_1                   = nan errno 0 (Success) except NONE
        bad value got sNaN expect qNaN
    scalb_1_snan                   = nan errno 0 (Success) except NONE
        bad value got sNaN expect qNaN
    nextafter_qnan_1               = nan errno 0 (Success) except NONE
        bad value got sNaN expect qNaN
    nextafter_snan_1               = nan errno 0 (Success) except NONE
        bad value got sNaN expect qNaN
    nextafter_1_qnan               = nan errno 0 (Success) except NONE
        bad value got sNaN expect qNaN
[... 66 lines cut ...]
    expm1_snan                     = nan errno 0 (Success) except NONE
        exceptions supported. expm1_snan returns NONE instead of FE_INVALID
    expm1_big                      = inf errno 0 (Success) except NONE
        exceptions supported. expm1_big returns NONE instead of FE_OVERFLOW
    expm1_negbig                   = -1 errno 0 (Success) except NONE
        exceptions supported. expm1_negbig returns NONE instead of FE_INEXACT
    fdim_snan_1                    = nan errno 0 (Success) except NONE
        exceptions supported. fdim_snan_1 returns NONE instead of FE_INVALID
    fdim_1_snan                    = nan errno 0 (Success) except NONE
        exceptions supported. fdim_1_snan returns NONE instead of FE_INVALID
    floor_snan                     = nan errno 0 (Success) except NONE
        exceptions supported. floor_snan returns NONE instead of FE_INVALID
    fma_big_big_1                  = inf errno 0 (Success) except NONE
        exceptions supported. fma_big_big_1 returns NONE instead of FE_OVERFLOW
    fma_big_negbig_1               = -inf errno 0 (Success) except NONE
        exceptions supported. fma_big_negbig_1 returns NONE instead of FE_OVERFLOW
    fma_small_small_0              = 0 errno 0 (Success) except NONE
        exceptions supported. fma_small_small_0 returns NONE instead of FE_UNDERFLOW
    fma_small_negsmall_0           = -0 errno 0 (Success) except NONE
        exceptions supported. fma_small_negsmall_0 returns NONE instead of FE_UNDERFLOW
    fma_1_1_snan                   = nan errno 0 (Success) except NONE
        exceptions supported. fma_1_1_snan returns NONE instead of FE_INVALID
    fmax_snan_snan                 = nan errno 0 (Success) except NONE
        exceptions supported. fmax_snan_snan returns NONE instead of FE_INVALID
```

keith-packard, 2024-12-30:
We're currently using superh bits built from crosstool-ng 1.25 which uses gcc 13.2.0. I suspect something changed since then...

olegendo, 2024-12-30:
> For SuperH, as per my digging into GCC, it seems like reversed NaN signalling bits and FP exceptions for soft-float is never handled properly in GCC. I don't understand how can CI work for now...
> 
> I just attempted to make a quick & dirty patch but it seems like there are still many test failures:
> 
> [FlyGoat/gcc@6970ecc](https://github.com/FlyGoat/gcc/commit/6970eccb94a2a867b4f80255dfd96b63c7c5e75c)
> 
> Maybe you or @olegendo know better than me :-)

Thanks for the patch proposal.

We already have an issue recorded for this in GCC's bugzilla:
https://gcc.gnu.org/bugzilla/show_bug.cgi?id=111814

The suggested starting point there is a bit different from your patch though.  I haven't been able to look at the issue myself.  If you get anything usable and working out of it, please keep me posted, I'll help to merge it upstream.

olegendo, 2024-12-30:
> We're currently using superh bits built from crosstool-ng 1.25 which uses gcc 13.2.0. I suspect something changed since then...

FYI, from GCC 15 on SH will use soft-fp instead of fdpbit

https://gcc.gnu.org/git/gitweb.cgi?p=gcc.git;h=e95512e2d5a317e8c043f232158df4b38186e51c

FlyGoat, 2024-12-30:
nvm I figured out that sh4 tests were never run in CI as qemu-system-sh4 is not present in current image.

FlyGoat, 2024-12-31:
> We're currently using superh bits built from crosstool-ng 1.25 which uses gcc 13.2.0. I suspect something changed since then...

Tried GCC 13.3.0, it's not working either. I'll just call it a day and upstream my current GCC fixes.

FlyGoat, 2024-12-31:
There is a funny error on MSP430:

```
2024-12-31T00:28:07.1721535Z 124/188 picolibc / printfm-tests                      FAIL               0.10s   exit status 1
2024-12-31T00:28:07.1727521Z ――――――――――――――――――――――――――――――――――――― ✀  ―――――――――――――――――――――――――――――――――――――
2024-12-31T00:28:07.1728373Z test 349 failed: expected "61" (2), got "161" (3)
2024-12-31T00:28:07.1731632Z test 349 failed: expected "61", got "161"
2024-12-31T00:28:07.1734696Z ――――――――――――――――――――――――――――――――――――――――――――――――――――――――――――――――――――――――――――――
```

keith-packard, 2024-12-31:
> There is a funny error on MSP430:
>
> ```
> 2024-12-31T00:28:07.1721535Z 124/188 picolibc / printfm-tests                      FAIL               0.10s   exit status 1
> 2024-12-31T00:28:07.1727521Z [...]

That's caused by printf lacking support for `%hhx` -- it's supposed to truncate the value 0x161 into a single byte. That's done with a cast through `unsigned char`, so perhaps the compiler is breaking that operation?

keith-packard, 2025-01-10:
m68k patches for gcc have been merged upstream

closed 2026-11-10 by keith-packard