picolibc/picolibc #939: _ALLOW_NEGATIVE_CTYPE_INDEX can be mismatched between library build and compiler settings

jamesl-dm, 2025-02-05:
Hi,

We ran into an issue with our builds where using ctype functions like isalpha() and toupper() would produce a linker error, complaining of an undefined reference to `_ctype_b'.

It seems like this is a result of our prebuilt picolibc binaries (picolibc-1.8.7-13.3.rel1) being built with -funsigned-char, whereas our codebase needs to be compiled with -fsigned-char. This leads to the ctype header redirecting to the wrong lookup table here:
https://github.com/picolibc/picolibc/blob/1e40e223ef6d5e2e28c730f2e8b4ee3c6f853865/newlib/libc/include/ctype.h#L225

I'm not sure if this is intended behaviour or not. I've never really thought about signed/unsigned settings affecting library builds, the way soft float / hard float does. I just assumed that everything would work no matter what settings was used.

Perhaps the macros can be reworked to cast the signed versions to unsigned on all builds?
Or would you recommend rebuilding with matching compiler settings?

Thanks,
James

keith-packard, 2025-02-05:
So this is all a kludge to work around application bugs. Applications often assume that the ctype functions take `char` parameters, (which is completely reasonable, given the use case). However, they do not. They take an `int` which must either be an unsigned char *or* EOF. To avoid crashing with an out-of-bounds array access when the application passes a signed char, the library create an array indexed from -127 to 255 on targets for which the default char type is signed.

It's all entirely unnecessary when applications follow the letter of the spec, but the spec is pretty terrible in this case, and having applications 'just work' even when they pass signed chars around seems like a useful feature to me.

First, I don't think you should have to change your compiler settings. The library API should work with either type.

That leaves us with a few options:

 1. Always allocate the larger array. That will ensure applications get this fail-safe behavior independent of the default char type.
 2. Always allocate the smaller array. This will save flash space and follow the letter of the spec.
 3. Select behavior when the library is built and record that in picolibc.h

I'm leaning towards option 1 here; it's a modest amount of flash space, and if you're really concerned about that, you can always use the smaller ctype implementation that encodes ASCII rules directly into the inline ctype functions. That only works when you don't enable extended charsets (other than ASCII and UTF-8).

jamesl-dm, 2025-02-05:
> They take an int which must either be an unsigned char or EOF. 

Yikes! I would never have guessed this in a thousand years.
I also like option 1 the best.

closed 2025-02-05 by keith-packard