picolibc/picolibc #956: sqrtf, rsqrt, reciprocal implementations in arm assembly

Kuratius, 2025-02-23:
I've written written an implementation for calculating the square root of the mantissa of a 32-bit float, including corresponding implementations for reciprocal square roots and the reciprocal of the mantissa.
I believe this to be significantly faster than the current implementation in Picolibc,
at least in the default rounding mode. On a Nintendo DSi I measured an execution time of 18 timer ticks for the assembly function, and 37 timer ticks for a C function that calls this routine internally to provide floating point emulation while adding special cases for things like NaN and inf.
I've measured the implementation used in blocksDS, which uses Picolibc I believe, and got an execution time of 109 ticks.

I've verified that the output of this function is within 1 ulp of the correct result, and that the correctly rounded result can be obtained by calculating the residual and and adding +1 to the result when appropriate. I can send you the code with my tests if it's helpful or open an issue for it on your repo.
I've also written a corresponding function for the reciprocal square root and the reciprocal of the mantissa with similar execution times. The function for the reciprocal of the mantissa is even faster I believe, however I haven't gotten around to doing extensive accuracy testing on it yet.

Would this be something you would be interested in for Picolibc?
I believe it can also be used to calculate the base logarithm to a good amount of precision, because essentially what it does it is it constructs an approximation to 1/sqrt(x) as a product of factors (1+2^-n), with the factor being chosen such that it gets you closer to 1/sqrt(x) (or 1/x for the reciprocal)

On the sqrt_asm the rounding can be fixed with    
 
```
    uint32_t new_mantissa=sqrt_asm(mantissa32);
    int64_t msquared=(int64_t)new_mantissa*new_mantissa;
    int64_t x0=mantissa32;
    x0<<=23;
    int64_t residual=x0-msquared;
    //residual>>=23;
    if (residual>new_mantissa)
        {
            new_mantissa+=1;
        }  
```

```
.syntax  unified
.global sqrt_asm
.arm
.balign 4
.type sqrt_asm, %function
sqrt_asm: //not safe to use with r0=0 
    //calculates sqrt of 9.23 fixed point number in the range [1,4)
    lsls r0, r0, #7      
    mov  r2, r0
    .irp x,1 ,2,3,4,5,6,7,8,9,10,11,12,13//further iterations are not necessary, x=0 adds extra range
    adds   r1,r0,r0,LSR #\x  
    addscc r1,r1,r1,LSR #\x
    movcc  r0,r1
    addcc  r2, r2,r2,LSR #\x
    .endr
    negs    r0, r0    
    umull  r1,r3, r0, r2
    adds   r0, r2,r3, lsr #1 
    lsrs r0, r0, #8
    bx lr

.global rsqrt_asm
.arm
.balign 4
.type rsqrt_asm, %function
rsqrt_asm: 
    //calculates rsqrt of 9.23 fixed point number in the range [1,4)
    lsls r0, r0, #7      
    mov  r2, #1<<31
    .irp x,1,2,3,4,5,6,7,8,9,10,11,12
    ADDS   r1,r0,r0,LSR #\x  
    ADDSCC r1,r1,r1,LSR #\x
    MOVCC  r0,r1
    ADDCC  r2, r2,r2,LSR #\x
    .endr
    lsr r0, r0, #2
    RSB r0, r0, #3<<30
    umull  r1,r3, r0, r2
    lsrs   r0, r3, #8
    bx lr

.global reciprocal_asm
.arm
.balign 4
.type reciprocal_asm, %function
reciprocal_asm: 
    //calculates reciprocal of 9.23 fixed point number in the range [1,2)
    lsls r0, r0, #8      
    mov  r2, #1<<31
    .irp x, 1,2,3,4,5,6,7,8 
    ADDS   r1,r0,r0,LSR #\x  
    MOVCC  r0,r1
    ADDCC  r2, r2,r2,LSR #\x
    .endr
    lsrs   r0,r0, #1
    negs    r0, r0 
    umull  r1,r3, r0, r2
    lsrs   r0, r3, #8
    bx lr
```

keith-packard, 2025-02-24:
Happy to add a faster sqrt for devices without an FPU. The IEEE standard requires correct rounding in all cases, along with NaN and INF support, but soft float mode on ARM only has round-nearest-even, so at least you don't need to deal with the other modes

Kuratius, 2025-02-28:
[sqrt-tests.zip](https://github.com/user-attachments/files/19029477/sqrt-tests.zip)
This contains a simple accuracy test that iterates over all mantissas for an even exponent and an odd one, in this case from [126<<23, 128<<23) .
 I ran it using qemu with the command
```
arm-linux-gnueabihf-gcc -static custom-sqrt.c sqrt.s -lm  && qemu-arm -L /usr/arm-linux-gnueabihf ./a.out
```
The "hf" suffix is necessary if you want to run the test in a few seconds, otherwise it takes somewhere around a minute or two.

Sqrt has the rather convenient quirk that ties are impossible if the input was a rational number, so you can just implement nearest instead of nearest-even rounding, as there are never any ties if you calculated all digits correctly.

The scaffolding I have here doesn't have support for subnormals, but it could be added without too much trouble I believe, and it shouldn't affect test accuracy as subnormal numbers would just get shifted into the range of a normalized mantissa before doing the calculation.

I don't know where you would prefer to put this, into its own function, or if you would just like to replace the mantissa calculation in an existing function. I would appreciate advice in that regard, otherwise you'd have to manually fix any PR I make.

[image]

Originally I took the idea for this method from Jan Vlientinck (see the last section here: https://www.pertinentdetail.org/sqrt ) who has written a variant that can be adapted to use a MUL instead of a UMULL, but his code seems to not have been fully tested or based on trial and error with a limited set of inputs. He also seems to not have been aware that he was calculating 1/sqrt(x), and it doesn't seem to have been written with ieee754 floats in mind.
I've tried making modifications to do the same here, and my conclusion is that using MUL is not worth it because it requires at least 3 extra iterations of the main loop, and 
```add Y, Y, #1``` 
this line in his implementation of the idea doesn't give you correct rounding for some 0.6 % of inputs because the calculation method doesn't give you the correct rounding digit in all cases, even if you change it to something like ```add Y,Y, #0x80```.

Kuratius, 2025-03-06:
A quick update, I went and tested a reciprocal sqrt with NaN/Inf support using this code. I had to downshift part of the calculation by 1 digit to prevent an overflow when rounding, but this only affects accuracy by very little (and you can technically prevent it by doing the rounding in 3 instead of 2 instructions).

Compared to (float)(1.0/sqrt((double) (float) x) ).
92 % of mantissas give an exact match using an add as rounding
8 % are off by at most 1 ulp
with more iterations in the main loop you can get 98 -99 % exact matches, but I don't think that would be worth it.

It is, however, significantly more accurate than doing 1.0f/sqrtf(x) with both the sqrt and the division in single precision; there the difference can be more than 1 ulp and only 62 % of tests give an exact match.
I'll see if I can find a way to make the rounding exact (or at least close to exact) without needing more iterations.
The rsqrtss instruction on x86 doesn't guarantee correct rounding.
I'm uncertain if the **r**sqrtf() function in C would need the guarantee that the single precision result is the same as the infinite precision result correctly rounded the way you have it with sqrtf(). If it doesn't then this would already be good enough.

Kuratius, 2025-03-09:
```static inline uint32_t mantissa_rsqrt(uint32_t x)
{
    uint32_t Y=x<<(0);
    uint32_t U=sqrt_core(x,1u<<30 )>>6;
    U |=1;
    uint32_t A_hi= ((uint64_t) U*U)>>32;
    uint32_t A_lo= U*U;
    uint32_t B_hi = ((uint64_t) Y*A_lo)>>32;
    uint64_t C= (uint64_t) Y*A_hi;
    uint64_t S=B_hi+C;
    uint32_t P=S>>32;
    uint32_t Q=1u<<(9);
    if (P >= Q)
    { 
       //count+=1;
       return  U>>1;
    } else {
        //count2+=1;
        return (U+1)>>1;
    }
}
```
this fixes the rounding to be 100 % identical to the double version without incurring extra iterations.

Kuratius, 2025-03-15:
@keith-packard I made a repository with my tests and different versions of the functions that may be applicable for certain architectures.
https://github.com/Kuratius/sqrtf-tests

SamB, 2026-03-05:
@Kuratius wrote:
> checking the carry flag from C is probably not possible, 

It might be interesting to check if https://gcc.gnu.org/onlinedocs/gcc/Integer-Overflow-Builtins.html helps? 

See also https://clang.llvm.org/docs/LanguageExtensions.html#multiprecision-arithmetic-builtins and https://clang.llvm.org/docs/LanguageExtensions.html#checked-arithmetic-builtins for the ones Clang knows about; I *think* they have everything but `__builtin_add_overflow_p`/`__builtin_sub_overflow_p`/`__builtin_mul_overflow_p` at the moment, though GCC could add more.

(Hopefully this won't be an an issue and, after they do memory-to-register conversion, both GCC and Clang will figure out on their *own* when you don't actually need the main results from the non-`_p` versions and skip computing them.)

It is unfortunate that the `__builtin_addc` and `__builtin_subc` functions aren't size-generic like `__builtin_add_overflow`/``__builtin_sub_overflow` are. (Well, maybe not *exactly* the same way: distinct types for each input and output would be excessive...)

Kuratius, 2026-03-05:
I've tried rewriting it using overflow builtins, the generic ones generate horrific assembly and the non-generic ones are still one instruction slower than the one in C where I used the sign bit instead of the carry flag.

There's also currently a regression in gcc that makes the C version take 5 instructions instead of the ideal 4, I've mentioned in the PR.
https://gcc.gnu.org/bugzilla/show_bug.cgi?id=123604

Another option could be using inline assembler like this 

-O3: https://godbolt.org/z/Esfx7Pb6Y
-Os: https://godbolt.org/z/rP7bWWaq4
 which the compiler can unroll or roll up again as it wants, but it wouldn't run on thumb-2 or riscv.