picolibc/picolibc #988: random.c: Signed integer overflow when using sanitizers

GalaxyShard, 2025-04-05:
```c
NEWLIB_THREAD_LOCAL long long _rand_next = 1;

long int
random (void)
{
  /* This multiplier was obtained from Knuth, D.E., "The Art of
     Computer Programming," Vol 2, Seminumerical Algorithms, Third
     Edition, Addison-Wesley, 1998, p. 106 (line 26) & p. 108 */
  _rand_next = _rand_next * __extension__ 6364136223846793005LL + 1;
  return (long int)((_rand_next >> 32) & RAND_MAX);
}
```

Currently the above function (in libc/stdlib/random.c) can trigger Clang's undefined behavior sanitizer as in the following example: `signed integer overflow: 6364136223846793006 * 6364136223846793005 cannot be represented in type 'long long'`.

keith-packard, 2025-04-05:
Sounds like this should use unsigned long long instead? Or, rather, uint64_t?

keith-packard, 2025-04-05:
Oh, and which clang flags should I enable to see the message?

GalaxyShard, 2025-04-05:
> Sounds like this should use unsigned long long instead? Or, rather, uint64_t?

I wasn't quite sure whether the code expects two's complement wrapping or not, since negative numbers would affect the multiplication. `random` and `rand` are supposed to return non-negative integers though.

> Oh, and which clang flags should I enable to see the message?

I believe compiling a C file like I've written below with `-fsanitize=undefined` should do the trick. It's a runtime error so it will need to be executed too.

This should reproduce it (I haven't tested this yet):

```c
#include <stdlib.h>
int main() {

    // _rand_next is 1

    (void)rand();

    // _rand_next should now be 1*6364136223846793005 + 1 = 6364136223846793006

    // overflow in this call by 6364136223846793006*6364136223846793005 + 1
    (void)rand();

    return 0;
}

PR #989 (2025-04-07): stdlib: use unsigned arithmetic in rand() to fix overflow
Closes #988

I checked newlib's source code and found that they use `unsigned long long`, which is what I changed the previous `long long` to here.
closed 2025-04-07 by keith-packard