picolibc/picolibc #1007: [newlib][stdio]: nano-vfscanf.c 、nano-vfprintf.c... :

kaishaoshao, 2025-04-18:
nano-vfprintf.c, nano-vfprintf.c, etc.   function parameter calls the underlying function _printf_float, _scanf_float, etc. function in the 1.8.1 (I use the version is 1.8.8) after the version of the deleted the struct _reent *data this form of the parameter, but when calling the but then filled it in again, which is confusing and produces compilation errors. What is the reason for doing this.
The specific errors are as follows:

``` 
stdio/nano-vfprintf.c:612:32: error: use of undeclared identifier 'data'
  612 |             n = _printf_float (data, &prt_data, fp, pfunc, &ap_copy);
      |                                ^
1 error generated.
...
stdio/nano-vfprintf_float.c:66:15: warning: declaration of 'struct _reent' will not be visible outside of this function [-Wvisibility]
   66 | __cvt (struct _reent *data, _PRINTF_FLOAT_TYPE value, int ndigits, int flags,
.....
stdio/nano-vfscanf.c:402:22: error: use of undeclared identifier 'rptr'
  402 |         ret = _scanf_float (rptr, &scan_data, fp, &ap_copy);
      |                            ^
1 error generated.```

keith-packard, 2025-04-18:
I'm planning to delete the whole legacy stdio implementation fairly soon; can you explain why you're still using it?

kaishaoshao, 2025-04-19:
> I'm planning to delete the whole legacy stdio implementation fairly soon; can you explain why you're still using it?

I'm currently using the printf in the tinystdio directory and I found that the system call function of the sdk that has been adapted to newlib doesn't work, so I'd like to try to use the printf series in the stdio directory to see if it's compatible.

keith-packard, 2025-04-19:
I've got it building again, but the nano-printf code in the legacy stdio directory has a lot of bugs, especially for floating point.
Give #1008 a try and see if that helps. But, I'd strongly encourage you to transition to the tinystdio code when you can; that's going to be a lot smaller and more conformant.

kaishaoshao, 2025-04-19:
> I've got it building again, but the nano-printf code in the legacy stdio directory has a lot of bugs, especially for floating point. Give [#1008](https://github.com/picolibc/picolibc/pull/1008) a [...]

Thank you