picolibc/picolibc #1009: Add .note.gnu.build-id to the linker script

nolange, 2025-04-24:
I would like picolibc.ld to be modified the following way:

```
	.boot_flash : {
		KEEP (*(.text.init.enter))
		KEEP (*(.data.init.enter))
		KEEP (*(SORT_BY_NAME(.init) SORT_BY_NAME(.init.*)))
	} >boot_flash AT>boot_flash :text_boot_flash

	.note.gnu.build-id  : {
		*(.note.gnu.build-id)
	} >boot_flash AT>boot_flash :text_boot_flash

	.text : {
		/* Dont define in .note.gnu.build-id section, so it wont be emitted if empty */
		PROVIDE_HIDDEN (__gnu_build_id_start = SIZEOF(.note.gnu.build-id) ? ADDR(.note.gnu.build-id) + 16 : 0);

		/* code */
		*(.literal.unlikely .text.unlikely .literal.unlikely.* .text.unlikely.*)
```

Motivation for the change:

A `.note.gnu.build-id` section allows the usual tools to get an unique ID of the build for identifying ELF files,
and the flat binaries will have the 16 byte ELF note header to spot it.

Various tools like debuggers already support automatically fetching missing binaries or debuginfo vie Build-ID

The firmware itself can define `extern const char __gnu_build_id_start[1];` and output the Build-Id for diagnostics. (ie. the developer can then easily locate the ELF binary + debuginfo).

The Build-Id should be placed as early as possible, that way a different firmware will be detected after reading / comparing a single page of flash.

Whether a Build-ID will be added will is controlled by the existing `--build-id` linker flag.

keith-packard, 2025-04-24:
I can see the value in having this available, but I'd be happier if it were in the .rodata section instead of .text -- that would have less impact on the memory layout, where sticking this in front of the text section could easily break things.

keith-packard, 2025-04-24:
oh, also what's with the +16 in the address computation? What does this value come from, is it necessary and should it be done in the application instead?

nolange, 2025-04-25:
> I can see the value in having this available, but I'd be happier if it were in the .rodata section instead of .text -- that would have less impact on the memory layout, where sticking this in front of the text section could easily break things.

Its relevant that its as early as possible. For placing stuff like vector tables at the front, you have `.text.init.enter` and `.data.init.enter`. Otherwise its just random what is placed first in `.text` ?

> oh, also what's with the +16 in the address computation? What does this value come from, is it necessary and should it be done in the application instead?

There is an 16-byte ELF Note Header before the payload (the hash itself).

``` c++
    typedef struct
    {
        uint32_t n_namesz;			/* Length of the note's name.  */
        uint32_t n_descsz;			/* Length of the note's descriptor.  */
        uint32_t n_type;			/* Type of the note. (NT_GNU_BUILD_ID) */
        char namez[4];              /* 4 bytes for GNU\0 */
    } Elf_GnuNoteHdr;
    typedef struct
    {
        Elf_GnuNoteHdr hdr;
        unsigned char buildid[20];
    } Elf_GnuBuildId;

    Elf_GnuBuildId build_id = {
        { sizeof(Elf_GnuBuildId::hdr.namez), sizeof(Elf_GnuBuildId::buildid), 3 },
        {} /* linker places hash here */
    };
```

It is not absolutely necessary to point directly to the payload, but I consider that more useful. Been some time till I toyed around with it, but back then g++ 5 did reliable link pointers to linker symbols at compile time. It did not work with added arithmetic.

The bigger idea is to have a header early in the binary (after vector table), containing the build-id hash and other info that can be easily used from the firmware itself as well as quickly transferred (over a slow UART via STM32 Bootloader mode) and parsed from the system communicating with the micro controller.

``` c++
struct FwHeader {
        unsigned other_meta_stuff;
        const unsigned char *pBuildId;
    };

__attribute__((section(".data.init.enter")))
static const struct FwHeader fullheader =
{
    2,
    __gnu_build_id_start /* + 16 would need runtime initialization */
};
```

keith-packard, 2025-04-25:
Yeah, the linker can't do arithmetic in a relocation, so you have to do the arithmetic in the linker script if you need a symbol initialized like this. But, you can just declare the note structure and associated types in the file and fetch from there. At least that wouldn't hide the magic '16' value inside the linker script.

In any case, the basic plan here looks sound after I spent some time noodling with options that didn't work.

nolange, 2025-04-25:
> Yeah, the linker can't do arithmetic in a relocation, so you have to do the arithmetic in the linker script if you need a symbol initialized like this. But, you can just declare the note structure [...]

sure, but I dont know where I would be interested in the elf note header. a symbol directly to the an uuid is more useful to me.
reaching far, lld for windows offers a [__buildid symbol](https://lld.llvm.org/windows_support.html#lld-flags), directly to the build id. That would naturally have some windows PE header infront of it, likely with different offset.

that magic offset is better tied to the linker (+ scripts) magic where the outputformat is known.
firmware is not in any way tied to ELF, the real form is a flat binary. so those headers are magic as well.

i know thats a bit pathological, since practically all toolchains use ELF. i am not going to argue further about it.