picolibc/picolibc #1015: Error building with clang

braw-lee, 2025-04-28:
I am getting this linking error while trying to build picolibc with clang for x86_64 using the script `do-clang-x86_64-configure`

```
[1/1324] Linking target semihost/machine/x86/bios
FAILED: semihost/machine/x86/bios
clang -target x86_64-none-elf -march=core2 -mfpmath=sse -msse2 -fno-pic -fno-PIE -static -nostdlib  -o semihost/machine/x86/bios semihost/machine/x86/bios.p/bios.S.o -Wl,--as-needed -Wl,--no-undefined -fuse-ld=lld -nostdlib -m32 -Wl,--build-id=none -T /path/to/picolibc/semihost/machine/x86/bios.ld
ld.lld: error: section .text at 0xFFFF0000 of size 0x10000 exceeds available address space
collect2: error: ld returned 1 exit status
clang: error: linker (via gcc) command failed with exit code 1 (use -v to see invocation)
ninja: build stopped: subcommand failed.
```

braw-lee, 2025-04-28:
Possible culprit?
https://github.com/picolibc/picolibc/blob/main/semihost/machine/x86/meson.build#L79-L82

keith-packard, 2025-04-28:
Hrm. This works for me on Debian. Which OS are you building on?

braw-lee, 2025-04-29:
debian
Actually it looks like a linker bug in (clang,lld)-15
I was able to build successfully with latest clang,lld i.e 21

Thank you!

closed 2025-04-30 by keith-packard