picolibc/picolibc #1030: -Wl,--gc-sections support detection is broken?

s-barannikov, 2025-05-08:
The support is checked as:
```meson
test_link_defaults = cc.get_supported_link_arguments(['-Wl,--gc-sections', '-Wl,--no-warn-rwx-segments'])
```
and this results in:
```
-----------
Command line: `<...>/bin/clang -nostdlib <...>/testfile.c -o <...>/output.exe -D_FILE_OFFSET_BITS=64 -O0
    -Werror=implicit-function-declaration -Wl,--fatal-warnings -Wl,--gc-sections` -> 1
stderr:
ld.lld: error: cannot find entry symbol start; not setting start address
clang: error: ld command failed with exit code 1 (use -v to see invocation)
-----------
Compiler for C supports link arguments -Wl,--gc-sections: NO 
Running compile:
Working directory:  <...>
Code:
 int main(void) { return 0; }
```

The link check cannot succeed because there are no crt files yet (nor there is a semihosting library FWIW).
It looks like `--gc-sections` is only used for building tests. It would be nice if tests could be built with this option to detect malformed linker scripts etc.

I'm not familiar with the meson build system to come up with a fix.

```
The Meson build system
Version: 1.3.1
```

s-barannikov, 2025-05-08:
Locally I work around this by appending `,--entry=_main` to the tested string. (Most targets won't need the underscore before `main`.)

keith-packard, 2025-05-09:
yeah, it was tricky to get this to work with meson, but I think I managed. Check out #1031 and see if that does what you want.

s-barannikov, 2025-05-09:
That does what I want and looks great, thanks!

closed 2026-11-10 by keith-packard