picolibc/picolibc #1036: [Question] How to disable malloc?

fifteenhex, 2025-05-20:
Bit of background:

I'm converting an m68k mac linux bootloader called EMILE over from makefiles with its own local malloc(),printf() etc to meson and picolibc.

The mac ROM provides disk access, memory management etc and as it's apparently still running when the bootloader is up[0] so it seems best to use its memory management instead of trying to work out where to put a heap and having malloc() in picolibc. The original baked in malloc()/free() in the bootloader are wrappers for the ROM function...

0 - You can move the mouse around while the bootloader is active and the point updates on the screen, so someone is refreshing the screen state. I guess it is using a timer interrupt or something to take back control periodically.

The question:

I want to disable any malloc() provided by picolibc and with a bit of grep I found there is `MALLOC_PROVIDED` which is there for vxWorks because it has it's own malloc().

But this is in the non-nano malloc so I also need `newlib-nano-malloc=false` in the meson config.

This results in this build error however:

```
[1303/1303] Generating subprojects/picolibc/newlib/libc_duplicates with a custom command
FAILED: subprojects/picolibc/newlib/libc_duplicates 
/home/runner/work/EMILE/EMILE/subprojects/picolibc/scripts/duplicate-names /usr/bin/nm subprojects/picolibc/newlib/libc.a subprojects/picolibc/newlib/libc_duplicates
Duplicate names in subprojects/picolibc/newlib/libc.a
00000000 D _dummy_mallocr
```

If I comment out `_dummy_mallocr` in the source it builds and the resulting binary is working and using the ROM malloc() wrapper.

Is there a better way to disable any malloc in picolibc? Maybe the build failure is because I'm not doing it right?

Thanks!

keith-packard, 2025-05-20:
There isn't any way (currently) to eliminate malloc from the build. However, you can easily replace it by including your own version in the link command before -lc. Use object files instead of libraries so calls from the C library will use your versions even if the application doesn't use them.

It is a bit fiddly, so perhaps it is time to add a configuration variable that removes malloc from the build.

keith-packard, 2025-05-20:
I've created PR #1038 which provides an option to disable the internal malloc implementation.

closed 2025-05-21 by fifteenhex