picolibc/picolibc #1043: _PICOLIBC_CTYPE_SMALL silently overridden for C++

rdiez, 2025-05-25:
Documentation page [Ctype in Picolibc](https://github.com/picolibc/picolibc/blob/main/doc/ctype.md) talks about an alternative implementation of the Ctype family of functions. It does not say what the default for _PICOLIBC_CTYPE_SMALL is, but it does state that the default changes to 1 when optimising for size (-Os). It also states that you can force an implementation by setting _PICOLIBC_CTYPE_SMALL yoursef.

In file `newlib/libc/include/ctype.h` , I found this code:

```
/*
 * The small ctype code does not support extended character sets. It also breaks
 * libstdc++'s ctype implementation, so just skip it for c++
 */
#if defined(__MB_EXTENDED_CHARSETS_NON_UNICODE) || defined (__cplusplus)
#undef _PICOLIBC_CTYPE_SMALL
#define _PICOLIBC_CTYPE_SMALL 0
#endif

```

The trouble is, no matter what the default is, or what the user chooses, _PICOLIBC_CTYPE_SMALL is silently overridden when building in C++ mode (and when __MB_EXTENDED_CHARSETS_NON_UNICODE is defined too).

Therefore, if your project contains both C and C++ files, then you may inadvertently end up with both implementation variants in the same executable, one algorithm-based and one table-based.

Or have I missed something?

rdiez, 2025-05-25:
First of all, thanks for the information.

However, the problem remains that you may get the 2 different Ctype implementations at the same time in your executable, causing unnecessary bloat.

Furthermore, if you write C code, and then rename all of your files from .c to .cpp, there is no reason for you to suspect that the Ctype implementation has changed behind your back, which may also have an impact on the runtime performance.

Instead of Picolibc silently overriding the user's choice, I think it would be best to fail with an #error. This way, there would be no surprises.

keith-packard, 2025-05-25:
yeah, it's unfortunate that C++ has this weird behavior. I think a better fix would be to allow use of the inline functions from C++ code and fix libstdc++ to only use the tables for the APIs which require it, that way you could get the smaller version all of the time. It's one of those 'someday we'll fix this' items though, and hasn't risen to the top of my queue.

keith-packard, 2025-05-26:
> > that way you could get the smaller version all of the time.
> 
> I am no expert on such libstdc++ APIs, but if your C++ code uses both some inline Ctype routines, and also some libstdc++ API which in turn uses the table-based Ctype implementation, wouldn't you end up having both implementations in your executable

Yes. Which is why I decided to just select the tables version for any C++ app. I have no idea how prevalent usage of the table-based ctype stuff is in C++; I don't use that language any more than absolutely necessary.

> Until everything is sorted out in the future, I would at least document this issue in https://github.com/picolibc/picolibc/blob/main/doc/ctype.md . If you are short on time, a link to this GitHub Issue would suffice.

Feel free to submit a PR with whatever wording you'd prefer; I'm out all week at a conference and won't have much spare time, I'm afraid.

closed 2026-11-10 by keith-packard