picolibc/picolibc #1048: Upgrade from 1.8.9-2 to 1.8.10 has failing tests

RossSmyth, 2025-06-23:
Hello! I am the maintainer for picolibc in nixpkgs.

I am upgrading from 1.8.9-2 to 1.8.10, and am seeing two test failures:
```log
 >  42/145 picolibc:test / atexit                   FAIL            0.04s   exit status 1
       >  43/145 picolibc:test / on_exit                  FAIL            0.03s   exit status 1
       >
       > Ok:                134
       > Expected Fail:     3
       > Fail:              2
       > Skipped:           6
```
I played around with some of the meson flags and was unable to make them go away.

This is the configure flags that are used:
```
mesonConfigurePhase flags: --prefix=/nix/store/m43gw587cs9gyz93bf4p5s6srm2fyyv3-picolibc-1.8.10-2 --libdir=/nix/store/m43gw587cs9gyz93bf4p5s6srm2fyyv3-picolibc-1.8.10-2/lib --libexecdir=/nix/store/m43gw587cs9gyz93bf4p5s6srm2fyyv3-picolibc-1.8.10-2/libexec --bindir=/nix/store/m43gw587cs9gyz93bf4p5s6srm2fyyv3-picolibc-1.8.10-2/bin --sbindir=/nix/store/m43gw587cs9gyz93bf4p5s6srm2fyyv3-picolibc-1.8.10-2/sbin --includedir=/nix/store/r3ls35cllygnja6rpmdkz965v5q455y4-picolibc-1.8.10-2-dev/include --mandir=/nix/store/m43gw587cs9gyz93bf4p5s6srm2fyyv3-picolibc-1.8.10-2/share/man --infodir=/nix/store/m43gw587cs9gyz93bf4p5s6srm2fyyv3-picolibc-1.8.10-2/share/info --localedir=/nix/store/m43gw587cs9gyz93bf4p5s6srm2fyyv3-picolibc-1.8.10-2/share/locale -Dauto_features=enabled -Dwrap_mode=nodownload --buildtype=plain -Dmultilib=true -Dsanitize-bounds=false -Dsanitize-trap-on-error=false -Dsanitize-undefined=false -Dsanitize-allow-missing=false -Dprofile=false -Danalyzer=false -Dassert-verbose=true -Dfast-strcmp=true -Dsplit-large-tests=false -Duse-stdlib=true -Dfreestanding=false -Dnative-tests=true -Dnative-math-tests=true -Dpicolib=false -Dsemihost=false -Dfake-semihost=false -Dspecsdir=/nix/store/r3ls35cllygnja6rpmdkz965v5q455y4-picolibc-1.8.10-2-dev/lib -Dtinystdio=false -Dio-c99-formats=true -Dio-long-long=false -Dio-pos-args=false -Dio-long-double=false -Dio-float-exact=true -Datomic-ungetc=true -Dposix-console=true -Dformat-default=double -Dprintf-aliases=true -Dio-percent-b=false -Dprintf-small-ultoa=true -Dprintf-percent-n=false -Dminimal-io-long-long=false -Dfast-bufio=false -Dio-wchar=false -Dstdio-locking=false -Dmb-capable=false -Dmb-extended-charsets=false -Dmb-ucs-charsets=auto -Dmb-iso-charsets=auto -Dmb-jis-charsets=auto -Dmb-windows-charsets=auto -Dpicocrt=false -Dpicocrt-enable-mmu=true -Dpicocrt-lib=true -Dpicoexit=true -Dinitfini-array=true -Dinitfini=false -Dcrt-runtime-size=false -Dnewlib-register-fini=true -Dnewlib-nano-malloc=true -Dnano-malloc-clear-freed=false -Dsingle-thread=true -Dthread-local-storage=picolibc -Dtls-model=global-dynamic -Dnewlib-global-errno=false -Derrno-function=auto -Dtls-rp2040=false -Dstack-protector-guard=auto -Dwant-math-errno=false -Dtests=true -Dtests-cdefs=false
```
These is the failure output. They aren't really that helpful though.
```log
picolibc>  41/166 picolibc:test / atexit                            FAIL            0.04s   exit status 1
picolibc> >>> UBSAN_OPTIONS=halt_on_error=1:abort_on_error=1:print_summary=1:print_stacktrace=1 PICOLIBC_TEST=1 PATH=/build/source/scripts:/nix/store/qxxhbjhda3sm3v4j571vp2rzngcl5mi3-meson-1.8.0/bin:/nix/store/1492q00cm64n0hs5966s8cqj6j0x5nxg-ninja-1.12.1/bin:/nix/store/1c8sm86wj45vwkb3ww2b870h9i9wna6r-patchelf-0.15.0/bin:/nix/store/pbqah1qk4b5y14fqinr1h8zvhqy71v81-gcc-wrapper-14.3.0/bin:/nix/store/sa7j7cddyblhcb3ch3ds10w7nw75yjj1-gcc-14.3.0/bin:/nix/store/mdmsnfcvxyk5ynz7nx8nhss1wig0gljx-glibc-2.40-66-bin/bin:/nix/store/psy9v2asypgl9ylg8cnzkixc7fv0snj0-coreutils-9.7/bin:/nix/store/cadx5p7c0i06gf6h84iw9mrhx56imbv0-binutils-wrapper-2.44/bin:/nix/store/z3za8hfc24wb117s50p8b10agjkgm039-binutils-2.44/bin:/nix/store/psy9v2asypgl9ylg8cnzkixc7fv0snj0-coreutils-9.7/bin:/nix/store/c14zwgl8hf1wm0izij2i16xvk8ak70cy-findutils-4.10.0/bin:/nix/store/ibx4jfwlhjg4g0s6rrxrpaxa3ka8ns4m-diffutils-3.12/bin:/nix/store/pr318zsl44jdwpk9wk0sdrn19b6in7ah-gnused-4.9/bin:/nix/store/bc6zxzjnkjp4r9nhz5imy3cypvdh6r4n-gnugrep-3.12/bin:/nix/store/nv3y7zb1cwz1h9qy7nwz0s54j8dl1kqj-gawk-5.3.2/bin:/nix/store/lp82dcnrzljyix6yigwzrlpr1smvpmb0-gnutar-1.35/bin:/nix/store/6ag5dhk7sma61p6vl0kazfmpbrq08nqh-gzip-1.14/bin:/nix/store/ykdv4id6893gmkqwdmbimq237c1xqvq7-bzip2-1.0.8-bin/bin:/nix/store/6bwp1y45zlyvpr4ja2sk1yi9v5mrs94x-gnumake-4.4.1/bin:/nix/store/00zrahbb32nzawrmv9sjxn36h7qk9vrs-bash-5.2p37/bin:/nix/store/c9xmgszbf6i4dfq9r953khk9d7fdqigw-patch-2.8/bin:/nix/store/ikfwx7kbwz9zr7fziiac7f57jgbh3bnv-xz-5.8.1-bin/bin:/nix/store/3pdmbqy86wsbjdazxv1n3vrmj60vn0ri-file-5.45/bin ASAN_OPTIONS=halt_on_error=1:abort_on_error=1:print_summary=1 MALLOC_PERTURB_=11 MSAN_OPTIONS=halt_on_error=1:abort_on_error=1:print_summary=1:print_stacktrace=1 MESON_TEST_ITERATION=1 /build/source/build/test/atexit
picolibc>
picolibc>  42/166 picolibc:test / on_exit                           FAIL            0.04s   exit status 1
picolibc> >>> MALLOC_PERTURB_=218 PICOLIBC_TEST=1 UBSAN_OPTIONS=halt_on_error=1:abort_on_error=1:print_summary=1:print_stacktrace=1 PATH=/build/source/scripts:/nix/store/qxxhbjhda3sm3v4j571vp2rzngcl5mi3-meson-1.8.0/bin:/nix/store/1492q00cm64n0hs5966s8cqj6j0x5nxg-ninja-1.12.1/bin:/nix/store/1c8sm86wj45vwkb3ww2b870h9i9wna6r-patchelf-0.15.0/bin:/nix/store/pbqah1qk4b5y14fqinr1h8zvhqy71v81-gcc-wrapper-14.3.0/bin:/nix/store/sa7j7cddyblhcb3ch3ds10w7nw75yjj1-gcc-14.3.0/bin:/nix/store/mdmsnfcvxyk5ynz7nx8nhss1wig0gljx-glibc-2.40-66-bin/bin:/nix/store/psy9v2asypgl9ylg8cnzkixc7fv0snj0-coreutils-9.7/bin:/nix/store/cadx5p7c0i06gf6h84iw9mrhx56imbv0-binutils-wrapper-2.44/bin:/nix/store/z3za8hfc24wb117s50p8b10agjkgm039-binutils-2.44/bin:/nix/store/psy9v2asypgl9ylg8cnzkixc7fv0snj0-coreutils-9.7/bin:/nix/store/c14zwgl8hf1wm0izij2i16xvk8ak70cy-findutils-4.10.0/bin:/nix/store/ibx4jfwlhjg4g0s6rrxrpaxa3ka8ns4m-diffutils-3.12/bin:/nix/store/pr318zsl44jdwpk9wk0sdrn19b6in7ah-gnused-4.9/bin:/nix/store/bc6zxzjnkjp4r9nhz5imy3cypvdh6r4n-gnugrep-3.12/bin:/nix/store/nv3y7zb1cwz1h9qy7nwz0s54j8dl1kqj-gawk-5.3.2/bin:/nix/store/lp82dcnrzljyix6yigwzrlpr1smvpmb0-gnutar-1.35/bin:/nix/store/6ag5dhk7sma61p6vl0kazfmpbrq08nqh-gzip-1.14/bin:/nix/store/ykdv4id6893gmkqwdmbimq237c1xqvq7-bzip2-1.0.8-bin/bin:/nix/store/6bwp1y45zlyvpr4ja2sk1yi9v5mrs94x-gnumake-4.4.1/bin:/nix/store/00zrahbb32nzawrmv9sjxn36h7qk9vrs-bash-5.2p37/bin:/nix/store/c9xmgszbf6i4dfq9r953khk9d7fdqigw-patch-2.8/bin:/nix/store/ikfwx7kbwz9zr7fziiac7f57jgbh3bnv-xz-5.8.1-bin/bin:/nix/store/3pdmbqy86wsbjdazxv1n3vrmj60vn0ri-file-5.45/bin ASAN_OPTIONS=halt_on_error=1:abort_on_error=1:print_summary=1 MSAN_OPTIONS=halt_on_error=1:abort_on_error=1:print_summary=1:print_stacktrace=1 MESON_TEST_ITERATION=1 /build/source/build/test/on_exit
picolibc>
picolibc>  43/166 picolibc:test / timegm                            OK              0.03s
```

I am unsure how familiar you are with Nix, but here is the derivation:
https://github.com/RossSmyth/nixpkgs/blob/5a50ac463d69763dee1de2677f083341d901e27f/pkgs/by-name/pi/picolibc/package.nix

It is mostly exposing the meson options are nix options. When built on Linux all of the options that are defaulted to `stdenv.hostPlatform.isNone` evaluate to false, for example https://github.com/RossSmyth/nixpkgs/blob/5a50ac463d69763dee1de2677f083341d901e27f/pkgs/by-name/pi/picolibc/package.nix#L28. And the opposite for the ones that are `!stdenv.hostPlatform.isNone`.

There aren't any patches or anything. Just the config stuff so people can customize it for their purposes.

Any help you can provide would be great.

RossSmyth, 2025-06-23:
Yeah just native x64 linux

keith-packard, 2025-06-24:
You cannot use -Dinitfini-array=true when testing the library. Disable that (as seen in the sample scripts/do-native-configure) and the tests pass.