picolibc/picolibc #1066: Testing NaN values

quic-k, 2025-07-24:
For math's tests, I am looking at this function [fmag_of_error](https://github.com/picolibc/picolibc/blob/main/newlib/libm/test/test.c#L256)
IIUC this function will signal if the 2 arguments passed are equal or not, if not, it will also return the largest bit that was different

But is this full proof when dealing with NaN values as there are multiple representations for NaN?

#### Example
`0x7FC00000` and `0xFFFFFFFF` are both NaN by [definition](https://en.wikipedia.org/wiki/NaN)
the exponent bits are all 1s and mantisa part is a non zero for both

but the above function will return error with magnitude 1
is this the intended behavior? Please correct me if I am wrong

#### Solution
a check that ensures both numbers are NaN and then returns early

@keith-packard

keith-packard, 2025-07-25:
> I am facing this issue for Hexagon (specifically these functions `strtof` `atoff`) couldn't figure out yet, the exact reason why different bits are generated, my guess is NAN macro is compile time constant whereas when NAN is calculated through strtof("NAN") it is calculated at runtime maybe we can look into this more when I push patches for Hexagon

`strtof` and friends all use a compile-time constant when they see the string 'nan'. I'm betting there's a difference between what that constant is and what the hardware actually generates when a hardware operation results in NaN. And those should really be the same, so in this case we probably do want to flag this as an error.

quic-k, 2025-07-28:
yes, you are correct, thanks
the macro value for NAN is 0x7FC00000 whereas the floating point multiplication of NAN with zero gives 0xFFFFFFFF
but do we want to flag this as an error?
does any spec or standard say that the input and output NaN values of an operation must have same bits?

keith-packard, 2025-07-28:
> yes, you are correct, thanks the macro value for NAN is 0x7FC00000 whereas the floating point multiplication of NAN with zero gives 0xFFFFFFFF but do we want to flag this as an error? does any spec or standard say that the input and output NaN values of an operation must have same bits?

It's definitely unusual for the toolchain to use a different value than the hardware generates. I don't think it violates any standard, but I'd go verify that the hardware really does generate 0xffffffff for invalid operations (like 0.0f/0.0f) and that you're not just seeing the compiler doing constant folding, then I'd see what __builtin_nanf("") and __builtin_nansf("") generate.

quic-k, 2025-07-29:
I verified the generated bits
__builtin_nanf("") generates the default bits for NAN, right now it generates 0x7FC00000
__builtin_nansf("") generates 0x7FA00000

this is a clang intrinsic so, it is resolved before any target specific stuff is done and thus it won't generate bits according to hardware

invalid operations in Hexagon are generating NAN with bits 0xFFFFFFFF, irrespective of the input operands
example:
```
r0 = 0x0
r1 = sfrecipa(r0,r0)
/* r1 is now 0xFFFFFFFF */

r0 = 0x7FC00000
r1 = 0x1 /* any value */
r1 = sfmpy(r0,r1) /* floating point multiply */
/* r1 is now 0xFFFFFFFF */
```

I tried getting different NAN bits for Hexagon with this
```
#ifdef __HEXAGON__
#define NAN __builtin_nanf("0xFFFFFFFF")
#endif
```
but it still turns off the sign bit, (0x7FFFFFFF), and the test case still fails
I checked the LLVM code, and they never enable the sign bit for __builtin_nanf, even if the payload string has it enabled
will check if that is the intended behavior or not

only when I hardcode NAN to 0xFFFFFFFF, the test case pass
```
#define 0xFFFFFFFF
```
but this is not the correct way

keith-packard, 2025-07-29:
Yeah, reading through the GCC source code, you can tell it that the NaN value should have all of the bits set, by adjusting the `canonical_nan_lsbs_set` value in the format. But that looks like a pain as there's no way to adjust the standard ieee_single_format value. Adding another is possible, but looks to be a bit of an effort as there are a few places which check explicitly for &ieee_single_format for some reason.

Check the existing code in tests/math_errhandling_tests.c -- the is_equal function returns true if both values are nan and both have the same signaling status.

PR #1069 (2025-08-05): Fix testsuite for NAN bit mismatch
Fixes #1066
closed 2025-08-05 by keith-packard