picolibc/picolibc #1067: Better implementation for the isleap() function

mariopal, 2025-07-28:
I propose changing the implementation of the following function in the file mktime.c:

```c
static inline int isleap (int y)
{
  // This routine must return exactly 0 or 1, because the result is used to index on __month_lengths[].
  // The order of checks below is the fastest for a random year.
  return y % 4 == 0 && (y % 100 != 0 || y % 400 == 0);
}
```

To this other, more optimal one:

```c
static inline int isleap (int y)
{
  // This routine must return exactly 0 or 1, because the result is used to index on __month_lengths[].
  return ((y * 1073750999UL) & 3221352463UL) <= 126976UL;
}
```

I have verified that the results are the same as with the original function, up to the year 102500, when the two functions differ. I believe the performance improvement for picolibc library is more important than covering such distant years.

keith-packard, 2025-07-28:
definitely a cute hack, but it doesn't seem worth the change given that it isn't correct over the full range of int values.

keith-packard, 2025-07-29:
Is the performance of isleap critical to your application? Because this version is going to generate more code...

Kuratius, 2025-07-29:
https://godbolt.org/z/b1hPschxq

Here's an example with branch hints for the compiler, in 75 % of cases the existing function can be made to return quicker than your implementation.

Kuratius, 2025-07-29:
Some sloppy python code suggests this is roughly correct
```
solar_year=365.2421881

def isleap(year):
    return (year % 4 == 0) and ((year % 100 != 0) or (year % 400 == 0))

leftover_days=0;
for year in range(0, 10**5):
    leftover_days+=solar_year-365;    

    if (isleap(year)):
        leftover_days-=1;

    if abs(leftover_days)>2 and  not isleap(year):
        print("year ", year, "is a leap year, but isleap(year)=",isleap(year) )
        if (leftover_days<0):
            leftover_days+=1
        else:
            leftover_days-=1;
            
        
print("Done", leftover_days)
```

closed 2026-11-10 by keith-packard