picolibc/picolibc #1091: Missing C11 threads.h

sgstreet, 2025-09-19:
Commit 3c026d9 removed threads.h with a comment of "not compatible with Zephyr's"  This header file is part of C11 standard library.  I'm using it to implement C11 threads in an embedded system environment.  IMHO deleting this header breaks C11 standard support to handle a downstream problem and I would like to see better solution that enables other use cases.  Can this commit be reverted?

keith-packard, 2025-09-20:
Hrm. threads.h is supposed to define a bunch of types which picolibc can't know because they depend upon the underlying implementation. That's what the original picolibc threads.h did in machine/_threads.h, and those types (of course) conflicted with Zephyr's definitions of the same types. They'd conflict with pretty much *any* existing implementation, so the question is how we provide the required header while still letting the underlying OS provide the implementation.

I wonder if we could have picolibc check for an externally provided header file that defined the required types and then
use that in threads.h in place of some stub types:

```
#if __has_include(<implementation/threads.h>)
#include <implementation/threads.h>
#else
typedef struct { int __dummy; } cnd_t;
...
#endif
void call_once(once_flag *flag, void (*func)(void));
...
```

Would that work in your environment?

sgstreet, 2025-09-20:
I'm not sure what the original Zephyr issue was, but your approach would work.  Here is my test patch to threads.h:

```
diff --git a/newlib/libc/include/threads.h b/newlib/libc/include/threads.h
index 00ff1c522..00a71dc17 100644
--- a/newlib/libc/include/threads.h
+++ b/newlib/libc/include/threads.h
@@ -28,9 +28,17 @@
 #define        _THREADS_H_
 
 #include <sys/cdefs.h>
-#include <machine/_threads.h>
 #include <time.h>
 
+#if __has_include(<machine/_threads.h>)
+#include <machine/_threads.h>
+#else
+typedef struct { int __dummy; } cnd_t;
+typedef struct { int __dummy; } mtx_t;
+typedef struct { int __dummy; } thrd_t;
+typedef struct { int __dummy; } tss_t;
+#endif
+
 _BEGIN_STD_C
 
 typedef void (*tss_dtor_t)(void *);
```
And here is my implementation of `machine/_threads.h`:

```
#ifndef __THREADS_H_
#define __THREADS_H_

#include <stdint.h>
#include <stdatomic.h>
#include <stdlib.h>

#include <linked-list.h>

#include <rtos/rtos-toolkit/scheduler.h>

#define __requires_exclusive(x)
#define __requires_unlocked(x)
#define __locks_exclusive(x)
#define __trylocks_exclusive(x,y)
#define __unlocks(x)

#define __THRD_STACK_SIZE 1024
#define __THRD_PRIORITY (SCHEDULER_NUM_TASK_PRIORITIES / 2)
#define __THRD_KEYS_MAX 8
#define __THRD_TSS_SIZE (sizeof(void *) * __THRD_KEYS_MAX)
#define __THRD_MARKER 0x137cc731UL

#define TSS_DTOR_ITERATIONS 5

#define ONCE_FLAG_INIT 0

typedef atomic_int once_flag;
typedef uintptr_t thrd_t;
typedef size_t tss_t;

enum {
	mtx_prio_inherit = 0x8
};

typedef struct mtx
{
	long value;
	struct futex futex;
	unsigned long type;
	long count;
} mtx_t;

typedef struct cnd
{
	struct mtx *mutex;
	unsigned long sequence;
	struct futex futex;
} cnd_t;

struct tss
{
	atomic_bool used;
	void (*destructor)(void *);
};

struct thrd
{
	int (*func)(void *);
	void *context;
	int ret;
	bool detached;
	bool terminated;
	thrd_t joiner;
	struct cnd joiners;
	struct linked_list thrd_node;
	void *tss[__THRD_KEYS_MAX];
	unsigned long marker;
	char stack[] __attribute__((aligned(8)));
};

typedef struct thrd_attr
{
	unsigned long flags;
	unsigned long priority;
	unsigned long affinity;
	size_t stack_size;
} thrd_attr_t;

#endif
```
As a further note, I needed to define away the lock annotations (i.e. __requires_exclusive) which were not present in `threads.h` before `1.8.10`. These appears come from newlib, but are not defined in picolibc's version `sys/cdefs.h`.  Here is the lock annotations from newlib's 'sys/cdefs.h`:

```
/*
 * Lock annotations.
 *
 * Clang provides support for doing basic thread-safety tests at
 * compile-time, by marking which locks will/should be held when
 * entering/leaving a functions.
 *
 * Furthermore, it is also possible to annotate variables and structure
 * members to enforce that they are only accessed when certain locks are
 * held.
 */

#if __has_extension(c_thread_safety_attributes)
#define	__lock_annotate(x)	__attribute__((x))
#else
#define	__lock_annotate(x)
#endif

/* Structure implements a lock. */
/* FIXME: Use __lockable__, etc. to avoid colliding with user namespace macros,
 * once clang is fixed: https://bugs.llvm.org/show_bug.cgi?id=34319 */
#define	__lockable		__lock_annotate(lockable)

/* Function acquires an exclusive or shared lock. */
#define	__locks_exclusive(...) \
	__lock_annotate(exclusive_lock_function(__VA_ARGS__))
#define	__locks_shared(...) \
	__lock_annotate(shared_lock_function(__VA_ARGS__))

/* Function attempts to acquire an exclusive or shared lock. */
#define	__trylocks_exclusive(...) \
	__lock_annotate(exclusive_trylock_function(__VA_ARGS__))
#define	__trylocks_shared(...) \
	__lock_annotate(shared_trylock_function(__VA_ARGS__))

/* Function releases a lock. */
#define	__unlocks(...)		__lock_annotate(unlock_function(__VA_ARGS__))

/* Function asserts that an exclusive or shared lock is held. */
#define	__asserts_exclusive(...) \
	__lock_annotate(assert_exclusive_lock(__VA_ARGS__))
#define	__asserts_shared(...) \
	__lock_annotate(assert_shared_lock(__VA_ARGS__))

/* Function requires that an exclusive or shared lock is or is not held. */
#define	__requires_exclusive(...) \
	__lock_annotate(exclusive_locks_required(__VA_ARGS__))
#define	__requires_shared(...) \
	__lock_annotate(shared_locks_required(__VA_ARGS__))
#define	__requires_unlocked(...) \
	__lock_annotate(locks_excluded(__VA_ARGS__))

/* Function should not be analyzed. */
#define	__no_lock_analysis	__lock_annotate(no_thread_safety_analysis)
```
I'm not sure what correct fix for the lock annotations should be?

keith-packard, 2025-09-21:
that all looks great. I'm not sure the system-specific header should be `machine/_threads.h` as there is a machine directory delivered with picolibc, but I don't have a great idea for an alternative name. Suggestions welcome.

As for the annotations; I'd have to study how they work in the compiler to understand whether picolibc should attempt to provide them or not. Again, suggestions welcome.

sgstreet, 2025-09-21:
> that all looks great. I'm not sure the system-specific header should be `machine/_threads.h` as there is a machine directory delivered with picolibc, but I don't have a great idea for an alternative name. Suggestions welcome.
> 
I still like machine/_threads.h as this matched newlib's layout.

> As for the annotations; I'd have to study how they work in the compiler to understand whether picolibc should attempt to provide them or not. Again, suggestions welcome.

The lock annotation are a clang thing.  Reviewing newlib's `sys/cdefs.h` you can find the usage of `__has_extension(c_thread_safety_attributes)` to enable/disable the lock annotations. I think it is sufficient to add this block (see previous comment) to picolibs `sys/cdefs.h`

keith-packard, 2025-09-22:
> I still like machine/_threads.h as this matches newlib's layout.

That's actually a good reason to avoid using this name -- people often end up with newlib headers in their search path, and I'd like to avoid a file with the same path but different contents. Plus, this file isn't machine (i.e. architecture) specific, it's OS specific.

> The lock annotations are a clang thing. Reviewing newlib's `sys/cdefs.h` you can find the usage of `__has_extension(c_thread_safety_attributes)` to enable/disable the lock annotations. I think it is sufficient to add this block (see previous comment) to picolibs `sys/cdefs.h`

And it looks like we can use the `__lockable__` variants to avoid user symbols as clang fixed that issue. Thanks for the reference.

sgstreet, 2025-09-27:
> That's actually a good reason to avoid using this name -- people often end up with newlib headers in their search path, and I'd like to avoid a file with the same path but different contents. Plus, this file isn't machine (i.e. architecture) specific, it's OS specific.

The file `machine/_threads.h` is not delivered as part of newlib.  When you try to include <threads.h> compilation fails if machine/_threads.h does not exit.  Which seems like reasonable behavior.

Do you have a recommendation for the new include path and filename?

keith-packard, 2025-09-28:
> The file `machine/_threads.h` is not delivered as part of newlib. When you try to include <threads.h> compilation fails if machine/_threads.h does not exit. Which seems like reasonable behavior.

Yeah, that does seem like the right behavior. I wonder if we should stick a check for _threads.h and `#define __STDC_NO_THREADS__ 1` when it isn't available?

> Do you have a recommendation for the new include path and filename? Maybe `bits/_threads.h`?

Thinking about it this afternoon, I wonder if we shouldn't use a directory name that just doesn't exist in picolibc at all. That way your OS environment could have a directory with that name and ensure that it could never collide with a picolibc or newlib filename? We could use that for other os-specific functionality, like the 'stat' structure, which is currently a hot mess.

Maybe `os/_threads.h`?

keith-packard, 2026-11-10:
PR #1464 added a test for an external <threads.h> and, if not present, defines `__STDC_NO_THREADS__`. So, when Zephyr provides its own threads.h, picolibc shouldn't define that, and the system will use the external file.

closed 2026-11-10 by keith-packard